freifunk-gluon/gluon #1843: feature request: rsyslog to suppress client MACs or any logs from hostapd

lrnzo, 2019-10-17:
Hello folks,

here at Freifunk Nordwest we would like to make use of the rsyslog feature and we are thinking about putting a default syslog server address in our site.conf, so people would just have to check the box in config mode and not do any research for the server address. Anyway. We wonder if there is a way to prevent rsyslog from transmitting the MACs of clients to the server, since

- there is no crypto and these are sensible date
- in my point of view these infos do not help with debugging, whereas debugging is the main purpose of having a central log server
- these infos massively increase the logfile on the server

the last two points could be solve on server side but the first one is most important and can only be fixed on client side, which is why i opened this issue.

so long.

rubo77, 2019-10-18:
Actually there IS crypto, if you use the fastd connection.

NeoRaider, 2019-10-18:
If hostapd already has such a feature, adding an option to enable it seems like a good idea to me.

An alternative would be to have the server sanitize the data. Because of the way batman-adv works, the MAC addresses of all clients connected to the mesh are known to every mesh node anyways, so protecting against eavesdropping during transport of the logs is more or less futile...

lrnzo, 2019-10-18:
oh, thanks for the hint. i looked up `/var/run/hostapd-phy0.conf` and it has `logger_syslog_level=2` in it. how do i restart the hostapd? `/etc/init.d/hostapd` does not exist and if i reboot the device, my changes to the config file are lost.

lrnzo, 2019-10-18:
hm, never mind. ill try building a firmware with logging defaults for `hostapd` adapted to my needs. one more question. what do i have to write into my site.conf in order to replace the `0.0.0.0` as logserver in the config mode? it would be nice if the node owner could just check the an have to research the correct server ip. I tried adding
```
	config_mode = {
		gluon_web_logging = {
			ip = {
				placeholder = '1.2.3.4',
			},
		},
```
which did not work.

closed 2019-10-19 by lrnzo