freifunk-gluon/gluon #1846: Move core-gluon packages into a separate repo

SvenRoederer, 2019-10-27:
I'm currently trying to reimplement some features of the [freifunk-berlin firmware-builder](https://github.com/freifunk-berlin/firmware), e.g. Patch-handling (https://github.com/freifunk-berlin/firmware/pull/714). Also our current way to create the list of packages for the image is not flexible enough.
To make use of gluon as a more generic "framework for creating OpenWrt-based firmwares for wireless mesh nodes" it would be great, to not have the core-gluon in the packages-directory. 

What are the chances to move this directory into a separate repo to reduce strict binding to gluon-based firmwares?

SvenRoederer, 2019-10-27:
Just seen a similar intention was raised in Issue https://github.com/freifunk-gluon/gluon/issues/1689 already.

NeoRaider, 2019-10-28:
At some point in the past, the Gluon base repo didn't contain any packages. We merged this eventually, as dependencies between the packages and Gluon's build system made the separation useless and often cumbersome (In the meantime, these dependencies have been reduced greatly, but there are still some left).

In particular, we want to ensure that a given commit of the base repo and site repo completely determines the used packages - it's bad enough that we need to update our `modules` file so often to get the latest OpenWrt etc. commits, we would really like to avoid having to do the same thing for the Gluon base packages.

While I wouldn't recommend it, using the Gluon base repo as an OpenWrt feed should work just fine (unless there are issues I'm not aware of), so this might be a solution to your problem.

(Note that Gluon is likely to move further away from a "regular" OpenWrt setup in the future - if I had the free time to work on this more actively, I'd love to get rid of UCI and netifd, as they aren't a good fit for the way Gluon manages its network configuration.)

SvenRoederer, 2020-02-18:
> While I wouldn't recommend it, using the Gluon base repo as an OpenWrt feed should work just fine (unless there are issues I'm not aware of), so this might be a solution to your problem.

My intention is more the other way around. 
The framework for building the gluon-images has some nice features and tools I'd to use in our Freifunk-Berlin build-framework. As we use a completely different internal setup than Gluon the core-packages are some "bloat" from the Freifunk-Berlin perspective.

SvenRoederer, 2020-05-26:
Just for reference: I made PR #1946 which shows roughly a way to generalize the build. Let's see when I find the time to rework it.