freifunk-gluon/gluon #1880: kernel running out of memory (unrelated to client count)

christf, 2019-11-24:
This may be similar to #1768 but I am raising this as a new issue because I see no connection to client count.

### Bug report

**What is the problem?**

* I have one client on a TL-WR740v4 that is active, another one is my phone which idles.
* After ~40 Minutes of my active client using the Freifunk connection at about top speed (~15-20MBit) throughput degrades, latency increases. Then below error messages are dumped into the log. 
* Then the device continues to function as normal, again reaching 20Mbit throughput.
* ps www does not show any memory hogs in userspace.
* netstat does not show any data sitting in queues
* load is starting out low - for 17Mbit at 1.2, gradually increasing over the first 40 minutes to 4. then dropping to 2 and rising again. Another 15 Minutes later the device reboots - unfortunatly I was unale to obtain logs from that.
* This is reproducible. With 1 client and that type of load from a big download I was able to re-trigger this after 40 minutes. 
* During this time roughly 4GB of data passed through the AP

**What is the expected behaviour?**
not running out of memory would be nice, see log

**Gluon Version:**

this is christf_next in https://github.com/christf/gluon/christf_next

**Custom patches:**

This is in the christf_next branch, so it uses:
* xlat464-clat
* wireguard
* ddhcp, l3roamd, mmfd
* To rule out triggers by cron scripts, I manually deactivated cron.

** Log **
```
Sun Nov 24 14:17:10 2019 daemon.err ddhcpd[2691]: 1 28 2 3 15 6 119 12 44 47 26 121 42 249 33 252 17 ]
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.541763] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.544377] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.debug kernel: [ 1858.563496] ieee80211 phy0: failed to copy skb for mesh0
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.569807] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.573605] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.576440] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.580530] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.584439] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.588477] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.592407] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.596401] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.600451] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.604382] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.608434] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.612379] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.616362] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.620418] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.625466] hostapd: page allocation failure: order:0, mode:0x2280020(GFP_ATOMIC|__GFP_NOTRACK)
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.632994] CPU: 0 PID: 1772 Comm: hostapd Not tainted 4.9.188 #0
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.639059] Stack : 80497522 00000035 00000000 00000000 00000000 00000000 00000000 00000000
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.647390]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.655723]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.664056]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.672389]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.680723]         ...
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.683153] Call Trace:
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.685447] [<8006ab2c>] 0x8006ab2c
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.688891] [<8006ab2c>] 0x8006ab2c
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.692361] [<800dead0>] 0x800dead0
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.695839] [<80460000>] 0x80460000
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.699306] [<800def04>] 0x800def04
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.702790] [<800de2b0>] 0x800de2b0
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.706259] [<80d9fe78>] 0x80d9fe78 [mac80211@80d80000+0x60040]
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.712157] [<80460000>] 0x80460000
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.715629] [<80112514>] 0x80112514
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.719106] [<80113194>] 0x80113194
Sun Nov 24 14:17:25 2019 kern.warn kernel: [ 1858.722579] [<80113260>] 0x80113260
[... 917 lines cut ...]
Sun Nov 24 14:17:26 2019 kern.warn kernel: [ 1863.269052] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:26 2019 kern.warn kernel: [ 1863.272984] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:26 2019 kern.warn kernel: [ 1863.276980] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:26 2019 kern.warn kernel: [ 1863.281030] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:26 2019 kern.warn kernel: [ 1863.284959] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:26 2019 kern.warn kernel: [ 1863.289009] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:26 2019 kern.warn kernel: [ 1863.292948] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:26 2019 kern.warn kernel: [ 1863.296938] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:26 2019 kern.warn kernel: [ 1863.300994] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:26 2019 kern.warn kernel: [ 1863.304927] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:26 2019 kern.warn kernel: [ 1863.308982] [nat46] Could not copy v4 skb
Sun Nov 24 14:17:26 2019 kern.debug kernel: [ 1863.312946] ieee80211 phy0: failed to copy skb for mesh0
Sun Nov 24 14:17:26 2019 kern.debug kernel: [ 1863.313037] ieee80211 phy0: failed to copy skb for mesh0
Sun Nov 24 14:17:26 2019 kern.debug kernel: [ 1863.313202] ieee80211 phy0: failed to copy skb for mesh0
Sun Nov 24 14:17:26 2019 kern.debug kernel: [ 1863.313255] ieee80211 phy0: failed to copy skb for mesh0
Sun Nov 24 14:17:26 2019 kern.debug kernel: [ 1863.313308] ieee80211 phy0: failed to copy skb for mesh0
Sun Nov 24 14:17:26 2019 kern.debug kernel: [ 1863.313387] ieee80211 phy0: failed to copy skb for mesh0
Sun Nov 24 14:17:26 2019 kern.debug kernel: [ 1863.313494] ieee80211 phy0: failed to copy skb for mesh0
Sun Nov 24 14:17:26 2019 kern.debug kernel: [ 1863.313601] ieee80211 phy0: failed to copy skb for mesh0
Sun Nov 24 14:17:26 2019 kern.debug kernel: [ 1863.313654] ieee80211 phy0: failed to copy skb for mesh0
Sun Nov 24 14:17:26 2019 kern.info kernel: [ 1863.324476] eth1: out of memory
Sun Nov 24 14:17:26 2019 daemon.err l3roamd[2465]: epoll error received on fd 8. Dumping fd: taskqueue.fd: 11 routemgr: 8 ipmgr: 7 icmp6: 12 icmp6.ns: 13 arp: 10 socket: 6, wifistations: 9 ... continuing
Sun Nov 24 14:17:26 2019 daemon.err l3roamd[2465]: handling netlink error-message: No buffer space available
Sun Nov 24 14:19:36 2019 daemon.err ddhcpd[2691]: 1 28 2 3 15 6 119 12 44 47 26 121 42 249 33 252 17 ]
``

christf, 2019-11-24:
This is what top looks like:
```
Mem: 24388K used, 3356K free, 68K shrd, 1808K buff, 4584K cached
CPU:   3% usr  29% sys   0% nic   4% idle   0% io   0% irq  62% sirq
Load average: 3.67 1.78 0.86 3/68 5401
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 5231     2 root     RW       0   0%  36% [kworker/0:5]
  999     2 root     SW       0   0%  23% [kworker/0:3]
 4900     2 root     SW       0   0%  12% [kworker/0:4]
 4680     2 root     SW       0   0%   8% [kworker/0:0]
    3     2 root     RW       0   0%   3% [ksoftirqd/0]
 3310     1 root     S     2696  10%   2% /usr/bin/respondd -d /usr/lib/respondd -p 1001 -g
 4200   788 root     S     1132   4%   1% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -
 1785     1 root     S     1548   6%   1% /usr/sbin/hostapd -s -P /var/run/wifi-phy0.pid -B
  704     1 root     R     1772   6%   1% /usr/sbin/haveged -w 1024 -d 32 -i 32 -v 1
 5235     2 root     SW       0   0%   1% [kworker/u2:3]
 4057  4049 root     R     1212   4%   1% top
  761     1 dnsmasq  S     1316   5%   0% /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg0141
 4201  4200 root     S     1204   4%   0% -ash
 4043   788 root     S     1132   4%   0% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -
 1084     1 root     S     1052   4%   0% /usr/sbin/babeld -D -I /var/run/gluon-babeld.pid -
 2711     1 root     S     1480   5%   0% /usr/sbin/ddhcpd -D -s 1 -b 2 -c local-node -i mmf
  847     1 root     S     1768   6%   0% /sbin/netifd
    1     0 root     S     1548   6%   0% /sbin/procd

```

mweinelt, 2019-11-24:
This looks like it's leaking in kernel space. You should probably try and rule out your kmods, especially the clat one.

christf, 2019-11-24:
I ran another series of tests like this:
* clat was loaded but not used.
* wget was executed 10 times om parallel to fetch data from an ipv6 host.  after 1h, 50 times parallel. After 3h I aborted the test. During this time, the load fluctuated - seemingly with the amount of parallel connections being active. With 60 wget connections, the load of the node approached 5.

My first test using clat was through a torrent with 240 parallel connections.
I just ran another test for >1h through clat with just 50 connections configured in ktorrent. The throughput is the same, however this time the node is stable.
While running the test /proc/net/nf_conntrack shows between 4k and 6k of entries.

Increasing the connection count in ktorrent after 1h runtime again

christf, 2019-11-24:
After running stable with reduced connection set but full bandwidth utilization for another hour, the proposal is to find a way to do clat without conntrack.

CodeFetch, 2020-01-14:
Hm... Maybe that's the the problem I've observed, too. I suspect conntrack is being used, because of the iptables [routing table lookup](https://github.com/freifunk-gluon/gluon/pull/1808/files#diff-1c584bef1b86f4267a137333b0234f2dR93). Because as far as I know the rule applies on a connection basis due to performance reasons.

Actually I can't believe it's due to conntrack. It's should not consume that much memory...
From the dmesg output it really looks like an OOM. Did you test it on a device with much more memory?

A bottleneck of the IPv4 to IPv6 translation mechanism is that the packet always needs to be copied. Thus I expect the issue to be there. It's very strange that this problem does not happen with less connections.

CodeFetch, 2020-01-14:
There was a point when I had to admit that 32MB are just not enough. Even the WLAN packet reassembly can consume more than 4MB in crowded environments as far as I remember... Memory usage can fluctuate too much without enough spare room. Still I think we have at least a bug in the WLAN implementation somewhere where packets reside too long before being freed (maybe WLAN packet reassembly). I also can't say I'm sure there is no bug in the CLAT.

CodeFetch, 2020-01-16:
@christf I'm 100% sure there is no memory leak in the nat46 module. 
All skbs are being freed (although kfree_skb is being used while dev_kfree_skb would be correct to my knowledge): 
https://github.com/ayourtch/nat46/blob/683fbd2b765506332a1af141545652bf58f03166/nat46/modules/nat46-netdev.c#L73

I think you really have to trace the slab allocations like we did in #1768.

Olf0, 2020-07-15:
When reading the OP of this issue, it documents the behaviour of my TP-Link TL-WR1043NDv1 (ath79:8/32) until a few weeks ago very well:
Throughout and latency becoming really bad after many hours to a few days uptime, then fluctuating (turning better and worse, but apparently never recovering to the full, original speed) for tens of minutes to hours, but ultimately always resulting in an auto-reset.  Then this loop started again with hours to days of regular runtime, until throughput & latency turn bad, then fluctuating, ...

But this behaviour / issue seems to be gone since my 1043NDv1 uses a Gluon 2020.2 based image!

mweinelt, 2022-03-27:
Two years have passed. If this is still a relevant issue it should be retested with a recent Gluon release.

closed 2022-03-27 by mweinelt

Olf0, 2022-03-27:
Sorry to state this so bluntly: Not cool to close an issue, only because nobody cared about it for two years.  There is no indication that this was fixed.  Plus, in the fragmented realm of Freifunk, some local groups are sure still using older gluon releases.

My FF-router has auto-update on; I will look what is currently running on it and try to retest (one must use it continuously for this, because after each auto-reset its is running fine for hours to days).

mweinelt, 2022-03-27:
Feel free to check and report whether this issue is still happening. But for us this issue is not actionable.