freifunk-gluon/gluon #1884: Node/Tunneldigger loses external (WAN) DNS resolver

RalfJung, 2019-11-27:
### Bug report

**What is the problem?**

We have now several times seen the following effect: a node suddenly drops offline, and when logging in via SSH, the log shows many of these errors:
```
Mon Nov 25 15:02:45 2019 daemon.err td-client: Hostname resolution timed out.
Mon Nov 25 15:02:45 2019 daemon.err td-client: Hostname resolution timed out.
Mon Nov 25 15:02:45 2019 daemon.err td-client: Hostname resolution timed out.
Mon Nov 25 15:02:45 2019 daemon.err td-client: Hostname resolution timed out.
```
However, the internet on the WAN port works fine, and a reboot fixes the problem. Seems like somehow the node "loses" its DNS server? Or maybe just tunneldigger is suddenly unable to resolve DNS names?

I have a hard time debugging DNS issues on Gluon devices as there are two dnsmasq running (I suspect one is a client on the WAN and one is the DNS server for nodes in the local mesh), and I don't know which one `nslookup` uses or how to even check the WAN-side DNS config. `/etc/resolv.conf` is not useful either. Is there a guide for debugging Gluon DNS issues?

**What is the expected behaviour?**
The node should be able to connect to the gateway as it has a working internet connection.

**Gluon Version:**
Seen with v2018.2.2 and v2019.1.

**Site Configuration:**
https://git.hacksaar.de/FreifunkSaar/gluon-site/tree/1.8.0-exp20191120

**Custom patches:**
None

NeoRaider, 2019-11-27:
The following information should give some insight:
- Output of `ifstatus wan` and `ifstatus wan6`, or just a full `ubus call network.interface dump`
- Contents of `/var/gluon/wan-dnsmasq/resolv.conf`

RalfJung, 2019-11-27:
Thanks, I will try to gather that data next time this happens.

NeoRaider, 2019-11-27:
There is also the `gluon-wan` command, which allows you to execute a command "in WAN context", so the command `gluon-wan nslookup heise.de` would use the DNS servers of the WAN interface instead of using the mesh interface.

RalfJung, 2019-11-27:
That sounds extremely useful indeed!

Is there a list of such debugging goodies somewhere?

RalfJung, 2019-11-29:
One of our users who experiences this problem fairly regularly was able to gather some data:

> Output of ifstatus wan and ifstatus wan6

```
{
        "up": false,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "proto": "dhcp",
        "device": "br-wan",
        "data": {

        }
}
{
        "up": false,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "proto": "dhcpv6",
        "device": "br-wan",
        "data": {

        }
}
```

> Contents of /var/gluon/wan-dnsmasq/resolv.conf

Seems to be empty.

NeoRaider, 2019-11-29:
Neither `wan` nor `wan6` have any configuration. It seems the node lost its DHCP lease. Does `logread` show anything related to the WAN interface when the issue occurs?

RalfJung, 2019-11-29:
The logread is full of this:
```
Thu Nov 28 21:54:26 2019 daemon.info td-client: Reinitializing tunnel context.
Thu Nov 28 21:54:26 2019 daemon.info td-client: Reinitializing tunnel context.
Thu Nov 28 21:54:26 2019 daemon.info td-client: Reinitializing tunnel context.
Thu Nov 28 21:54:26 2019 daemon.info td-client: Reinitializing tunnel context.
Thu Nov 28 21:54:26 2019 daemon.err td-client: No suitable brokers found. Retrying in 5 seconds
Thu Nov 28 21:54:31 2019 daemon.info td-client: Performing broker selection...
Thu Nov 28 21:54:34 2019 daemon.err td-client: Hostname resolution timed out.
Thu Nov 28 21:54:34 2019 daemon.err td-client: Hostname resolution timed out.
Thu Nov 28 21:54:34 2019 daemon.err td-client: Hostname resolution timed out.
Thu Nov 28 21:54:34 2019 daemon.err td-client: Hostname resolution timed out.
Thu Nov 28 21:54:35 2019 daemon.info td-client: Reinitializing tunnel context.
Thu Nov 28 21:54:35 2019 daemon.info td-client: Reinitializing tunnel context.
Thu Nov 28 21:54:35 2019 daemon.info td-client: Reinitializing tunnel context.
Thu Nov 28 21:54:35 2019 daemon.info td-client: Reinitializing tunnel context.
Thu Nov 28 21:54:39 2019 daemon.err td-client: Hostname resolution timed out.
Thu Nov 28 21:54:39 2019 daemon.err td-client: Hostname resolution timed out.
Thu Nov 28 21:54:39 2019 daemon.err td-client: Hostname resolution timed out.
Thu Nov 28 21:54:39 2019 daemon.err td-client: Hostname resolution timed out.
```
That seems to already push everything useful out of the log before the bug is noticed.

RalfJung, 2019-12-09:
We had that case again with a very differently-looking status:
<details>

```
### WAN network status:
{
    "up": true,
    "pending": false,
    "available": true,
    "autostart": true,
    "dynamic": false,
    "uptime": 254114,
    "l3_device": "br-wan",
    "proto": "dhcp",
    "device": "br-wan",
    "metric": 0,
    "dns_metric": 0,
    "delegation": true,
    "ipv4-address": [
        {
            "address": "192.168.1.100",
            "mask": 24
        }
    ],
    "ipv6-address": [

    ],
    "ipv6-prefix": [

    ],
    "ipv6-prefix-assignment": [

    ],
    "route": [
        {
            "target": "0.0.0.0",
            "mask": 0,
            "nexthop": "192.168.1.2",
            "source": "192.168.1.100\/32"
        }
    ],
    "dns-server": [

    ],
    "dns-search": [

    ],
    "inactive": {
        "ipv4-address": [

        ],
        "ipv6-address": [

        ],
        "route": [

        ],
        "dns-server": [
            "192.168.1.2"
        ],
        "dns-search": [

        ]
    },
    "data": {
        "hostname": "ffsaar-weber-und-nauert",
        "leasetime": 7200,
        "ntpserver": "192.168.1.2"
    }
}
{
    "up": true,
    "pending": false,
    "available": true,
    "autostart": true,
    "dynamic": false,
    "uptime": 254111,
    "l3_device": "br-wan",
    "proto": "dhcpv6",
    "device": "br-wan",
    "ip6table": 1,
    "metric": 0,
    "dns_metric": 0,
    "delegation": true,
    "ipv4-address": [

    ],
    "ipv6-address": [
        {
            "address": "2003:db:c710:2400:80c6:69ff:fe1f:bee8",
            "mask": 64,
            "preferred": 0,
            "valid": 14316
        },
        {
            "address": "fd1e:2045:8fe2::80c6:69ff:fe1f:bee8",
            "mask": 64,
            "preferred": 604716,
            "valid": 2591916
        }
    ],
    "ipv6-prefix": [

    ],
    "ipv6-prefix-assignment": [

    ],
    "route": [
        {
            "target": "2003:db:c710:2400::",
            "mask": 64,
            "nexthop": "::",
            "metric": 256,
            "valid": 14316,
            "source": "::\/0"
        },
        {
            "target": "::",
            "mask": 0,
            "nexthop": "fe80::209:4fff:feb3:d40c",
            "metric": 512,
            "valid": 1716,
            "source": "::\/0"
        }
    ],
    "dns-server": [

    ],
    "dns-search": [

    ],
    "inactive": {
        "ipv4-address": [

        ],
        "ipv6-address": [

        ],
        "route": [

        ],
        "dns-server": [
            "fe80::209:4fff:feb3:d40c"
        ],
        "dns-search": [

        ]
    },
    "data": {
        "passthru": "00170010fe8000000000000002094ffffeb3d40c001f0010fe8000000000000002094ffffeb3d40c"
    }
}
### WAN DNS:
nameserver fe80::209:4fff:feb3:d40c%br-wan
nameserver 192.168.1.2 
```

</details>

Logread is the same as before.

NeoRaider, 2019-12-10:
It would be interesting to know if *nslookup* and *ping* are still working under `gluon-wan` in this state.

RalfJung, 2019-12-10:
For the cases where DHCP was lost entirely, the user reported a crashing DHCP server on their pi-hole, so that might explain why the lease was gone. This does not explain the other occurrences of the problem, though.

Looks like we now had one case where restarting tunneldigger helped, and no full reboot was necessary. tunneldigger uses [a vendored version of libasyncns](https://github.com/wlanslovenija/tunneldigger/tree/master/client/libasyncns) for DNS resolution. Maybe that does not realize that the DNS server changed, and so DNS resolution fails?

RalfJung, 2019-12-10:
Seems like libasyncns in the end boils down to `getaddrinfo` (it's async because it has a threadpool to actually do those requests). And I assume `getaddrinfo` should work fine even when the DNS server changes? (Also it seems unlikely that the actual server changed, but DHCP might refresh some things or so? The entire thing is very odd.)

NeoRaider, 2019-12-10:
Tunneldigger only sees 127.0.0.1 as its resolver, that never changes (OpenWrt always handles DNS using a local dnsmasq instance). We use an iptables hack to forward all DNS packets from processes in the `gluon-mesh-vpn` group to an alternative dnsmasq instance running on port 54, which uses `/var/gluon/wan-dnsmasq/resolv.conf` as its upstream config. `gluon-wan` simply runs a process in the `gluon-mesh-vpn` group.

RalfJung, 2020-01-29:
> It would be interesting to know if nslookup and ping are still working under gluon-wan in this state.

`gluon-wan nslookup www.heise.de` does not work in such a state. `ping` we didn't try yet; we'll add it to the script.

NeoRaider, 2020-05-30:
Is this still reproducible with the latest v2020.1.x?

RalfJung, 2020-05-30:
We are just starting to ship that update on our experimental branch, so I can't tell yet.

rotanid, 2020-12-14:
@RalfJung any updates on this in newer Gluon releases?

RalfJung, 2020-12-14:
I have not seen this specific issue in a while, but I also have not been specifically looking for it.

rotanid, 2020-12-14:
@RalfJung feel free to re-open the issue if you encounter it specifically and/or can contribute something new to it

closed 2020-12-14 by rotanid