freifunk-gluon/gluon #1893: Docs: gluon-radv-filterd: unclear
“Local” routers

rubo77, 2019-12-22:
This section is really unclear and the first sentence is not finished:

> The package has functionality to select “local” routers, i.e. those connected via cable or WLAN instead of via the mesh (technically: appearing in the transtable_local), a fake TQ of 512 so that they are always preferred. However, if used together with the gluon-ebtables-filter-ra-dhcp package, these router advertisements are filtered anyway and reach neither the node nor any other client. You currently have to disable the package or insert custom ebtables rules in order to use local routers.

Jan das mal bitte jemand auf Deutsch erklären, dann kann ich einen pr erstellen, der das richtig stellt

NeoRaider, 2019-12-26:
Hmm, the intent is clear, but the grammar is off... maybe something like:

> "Local" routers that are connected to the client interface (via cable or WLAN) instead of via the mesh (technically: appearing in the transtable_local) are taken into account with a fake TQ of 512, so that they are always preferred.

Does this sound clearer?

I think the part about *gluon-ebtables-filter-ra-dhcp* should be reworded as well - using custom ebtables rules doesn't make much sense. Maybe we should just say that the use of local routers is not possible when that package is installed?

rubo77, 2019-12-27:
What kind of local routers?
What would you use such local routers for?

NeoRaider, 2019-12-27:
This would apply to Gluon-based supernodes, or supernode-less meshes with internet gateways connected to some nodes.

rotanid, 2019-12-28:
@rubo77 does this help you to finalize it?

rubo77, 2019-12-29:
Yes, i will make a pr

PR #1900 (2020-01-12): docs: gluon-radv-filterd: enhance documentation with example
closes #1893
closed 2019-12-29 by rubo77