freifunk-gluon/gluon #1898: "gluon-reload" does not reconfigure RAs

RalfJung, 2019-12-29:
## Bug report

I am doing first experiments with multi-domain support following [this](https://gluon.readthedocs.io/en/latest/features/multidomain.html). After setting up the domain, I used these instructions to configure my node:
```
uci set gluon.core.domain="test"
gluon-reconfigure
gluon-reload
```

**What is the problem?**
After doing that, the node still sends RAs for the old network, which are now wrong. I had to do `reboot` to actually properly configure the node for the new domain.

**What is the expected behaviour?**
The way I understood to the docs, `gluon-reload` should update all settings and programs accordingly with no reboot needed.

**Gluon Version:**
v2019.1

**Site Configuration:**
https://git.hacksaar.de/FreifunkSaar/gluon-site/tree/d882a49d3dd024f05248957291cb3955a4f57504

**Custom patches:**
Just swapped out `openwrt/package/base-files/files/etc/banner`.

RalfJung, 2019-12-29:
In fact, it's not just RAs that don't work properly.  DNS (`dig ralfj.de`) and IPv4 connectivity (`ping 1.1.1.1`) on a client attached to that node also did not work until a reboot.

NeoRaider, 2019-12-29:
Hmm, maybe we should just revert recommending to use `gluon-reload`, as it is a very hacky solution with very limited scope. I'm not even sure if it works well enough for the hoodselector...

NeoRaider, 2019-12-29:
Okay, there is definitely a restart of uradvd missing from `reload.d`, but I'm not sure what is wrong with DNS and IPv4, as `gluon-reload` does run `/etc/init.d/network restart`.

NeoRaider, 2019-12-29:
Hmm, gluon-ebtables might need a restart, too... and possibly other initscripts.

I might have an idea how to improve the situation, but this will require a procd patch - until then, rebooting is the way to go.

mweinelt, 2020-02-10:
I did a revert of the documentation commit, so we don't have it in v2020.1.

2tata, 2020-04-11:
@NeoRaider how I can proper solve this problem? At first I will add uradvd and gluon-ebtables to the reload scripts.

NeoRaider, 2020-04-11:
@2tata Sorry, I had started looking into this, but then forgot about it. My idea was to add a "soft reboot" feature to procd that will stop and then restart all system services, as it is done for a regular reboot, but without actually rebooting the kernel, and without remounting the filesystems (so /tmp contents etc. are preserved).

2tata, 2020-04-11:
hm, could be a way. The reason why I have implemented gluon-reload is to safe time between domain switching. With an reboot also decompressing image file and so on will be proceeded which is not really necessary. So the goal should be a small gap of time on switching the domain. Otherwise we could simply reboot the device.

NeoRaider, 2020-05-29:
This should be fixed with 22847e49867f5ff40d02b831a7bec9bbcfc962ba. Please reopen if there `gluon-reload` is still making trouble.

closed 2020-05-29 by NeoRaider

mweinelt, 2020-05-29:
What about the documentation revert in 2c14703? Should we revert that again?

NeoRaider, 2020-05-29:
I generally have mixed feelings about `gluon-reload`. We should definitely document that `gluon-reload` exists, but a good old uci commit + reboot still seems like the more robust alternative if a flash write and a slightly longer down time is acceptable for a domain switch.