freifunk-gluon/gluon #1902: logread -r gives false return-code

nudelkiste, 2020-01-01:
what I am doing:

> root@ffsaar-HammelsbergVa:~# logread -?
>     -r    <server> <port>    Stream message to a server
> root@ffsaar-HammelsbergVa:~# logread -r syslog.ffsaar 515
> root@ffsaar-HammelsbergVa:~# echo **$?**
> **0**
> root@ffsaar-HammelsbergVa:~#

-> I´d expect that everything went fine and my log was perfectly streamed to the server, but:

> root@ffsaar-HammelsbergVa:~# logread -l 1
> Wed Jan  1 21:52:47 2020 daemon.info logread[3904]: **failed** to send log data to syslog.ffsaar:515 via tcp
> root@ffsaar-HammelsbergVa:~# 

If there is some trouble sending the data, then the return-code of **logread -r** should be anything but "0", right?

ENV:
OpenWrt 18.06-SNAPSHOT, r7835+25-89808e211c
gluon-ffsaar 1.8.0 / gluon v2019.1

blocktrron, 2020-01-02:
Strictly speaking, this is not a Gluon specific bug but rather a upstream one in ubox. So the upstream Bugtracker is the better place for this issue.

nudelkiste, 2020-01-02:
not shure how to find the "upstream bugtracker"?
Hopefully I did right by creating:
https://bugs.openwrt.org/index.php?do=details&task_id=2711

closed 2020-01-02 by rotanid