freifunk-gluon/gluon #1906: add Support for TP-Link WA801ND v5.0

EvolverDE, 2020-01-05:
Since about 1 year this device laying here and waiting to be used as a freifunk-router. When are you finally creating an update for this?

rubo77, 2020-01-05:
It is supported by Openwrt, but it is absolutely not recommended,  with just 4MB and 32MB Flash, see: https://openwrt.org/toh/tp-link/tl-wa801nd

If you want to use a device for Freifunk, you should get one with min. 8MB RAM. for example the tp-link-tl-wa850re: 

https://geizhals.de/tp-link-tl-wa850re-a1238590.html?hloc=de

Or look at Ebay Kleinanzeigen, there are lots of used devices for 15€ with even 5 and 2.4 GHz

mweinelt, 2020-01-05:
> TL-WA801ND | v5 | MediaTek MT7628NN | 580 | 8 | 64

Get your facts straight. It's just that nobody with the device bothered to get it supported in Gluon thus far. Maybe you'd like to give it a try? https://gluon.readthedocs.io/en/v2019.1.x/dev/hardware.html

EvolverDE, 2020-01-06:
no, i dont have the skills for that.

heini66, 2020-01-06:
atm just supported in owrt trunk... build gluon with openwrt master or wait until gluon supports the current owrt master.

nalxnet, 2020-01-10:
I'll be receiving two of these next week and will try the integration when I'll find some spare time.

closed 2020-01-18 by blocktrron