freifunk-gluon/gluon #1917: build: Building a single package causes all dependencies to be checked

lemoer, 2020-01-11:
When a build for a single package is invoked, the whole dependencies are checked, which extends the build time from <1s to 26s on my machine:
``` 
lemoer@orange ~/d/f/g/gluon> cd openwrt; time make GLUON_TARGET=x86-generic GLUON_DEPRECATED=0 GLUON_DO_NOT_MINIFY=1 package/gluon-status-page/compile; cd ..
Collecting package info: done
 make[1] package/gluon-status-page/compile
 make[2] -C package/libs/toolchain compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/gluon/net/sse-multiplex compile
 make[2] -C package/utils/lua host-compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/packages/lang/luasrcdiet host-compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/gluon-neighbour-info compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/gluon/libs/libplatforminfo compile
 make[2] -C package/libs/libjson-c compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/gluon/net/respondd compile
 make[2] -C package/utils/lua compile
 make[2] -C package/libs/libubox compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/gluon/libs/lua-hash compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/gluon/libs/lua-jsonc compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/gluon/libs/lua-platform-info compile
 make[2] -C package/system/uci compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/gluon/libs/lua-simple-uci compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/gluon/admin/pretty-hostname compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/packages/lang/lua-cjson host-compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/gluon-site compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/libgluonutil compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/packages/lang/luabitop compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/packages/lang/luaposix compile
 make[2] -C package/firmware/linux-firmware compile
 make[2] -C package/firmware/prism54-firmware compile
 make[2] -C package/kernel/linux compile
 make[2] -C package/network/utils/iptables compile
 make[2] -C package/system/ubus compile
 make[2] -C package/network/config/firewall compile
 make[2] -C package/network/config/vxlan compile
 make[2] -C package/network/ipv6/odhcp6c compile
 make[2] -C package/libs/libnl-tiny compile
 make[2] -C package/network/utils/iwinfo compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/gluon-core compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/gluon-respondd compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/gluon-web host-compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/gluon-web compile
 make[2] -C package/libs/mbedtls compile
 make[2] -C package/libs/ustream-ssl compile
 make[2] -C package/network/services/uhttpd compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/gluon-status-page compile
20.23user 4.27system 0:23.73elapsed 103%CPU (0avgtext+0avgdata 118208maxresident)k
8inputs+5568outputs (0major+1537679minor)pagefaults 0swaps
```
This seems unnecessary in most of the cases to me.

Two ideas to improve the workflow here:
1. Maybe we can introduce a `GLUON_SKIP_DEPENDENCY_REBUILD` flag?
2. Another idea would be, that we expose the openwrt make targets to our make and add some additional targets. Eg: ```make package/PKG_NAME/build``` to build the package with dependencies and ```make package/PKG_NAME/rebuild``` to build the package without dependencies.

I would personally prefer option 2. What do you think?

SvenRoederer, 2020-01-11:
As you change to the openwrt directory you are using the OpenWrt Makefile already. As far sa I've seen, this dependency check belongs to OpenWrt.

lemoer, 2020-01-11:
Of course... Can you elaborate on your suggestion?

lemoer, 2020-01-11:
I found the location in openwrt where package dependencies could be deactivated:
```
diff --git a/scripts/package-metadata.pl b/scripts/package-metadata.pl
index 5abc1bdef3..c384f3d42f 100755
--- a/scripts/package-metadata.pl
+++ b/scripts/package-metadata.pl
@@ -419,16 +419,16 @@ sub gen_package_mk() {
                                # Filter out self-depends
                                my @vdeps = grep { $srcname ne $_->{src}{name} } @{$vpkg_dep};
 
-                               foreach my $vdep (@vdeps) {
-                                       my $depstr = sprintf '$(curdir)/%s/compile', $vdep->{src}{path};
-                                       if (@vdeps > 1) {
-                                               $depstr = sprintf '$(if $(CONFIG_PACKAGE_%s),%s)', $vdep->{name}, $depstr;
-                                       }
-                                       my $depline = get_conditional_dep($condition, $depstr);
-                                       if ($depline) {
-                                               $deplines{''}{$depline}++;
-                                       }
-                               }
+                               #foreach my $vdep (@vdeps) {
+                               #       my $depstr = sprintf '$(curdir)/%s/compile', $vdep->{src}{path};
+                               #       if (@vdeps > 1) {
+                               #               $depstr = sprintf '$(if $(CONFIG_PACKAGE_%s),%s)', $vdep->{name}, $depstr;
+                               #       }
+                               #       my $depline = get_conditional_dep($condition, $depstr);
+                               #       if ($depline) {
+                               #               $deplines{''}{$depline}++;
+                               #       }
+                               #}
                        }
 
                        my $config = '';

```

It is used in `include/topdir.mk`, you can invoke it manually using:
``` shell
./scripts/package-metadata.pl mk tmp/.packageinfo
```

lemoer, 2020-01-11:
Deactivating the regeneration of the package dependencies gives another few seconds
```
diff --git a/include/toplevel.mk b/include/toplevel.mk
index e744447b2f..4d2aba9d41 100644
--- a/include/toplevel.mk
+++ b/include/toplevel.mk
@@ -82,15 +82,15 @@ _ignore = $(foreach p,$(IGNORE_PACKAGES),--ignore $(p))
 prepare-tmpinfo: FORCE
        @+$(MAKE) -r -s staging_dir/host/.prereq-build $(PREP_MK)
        mkdir -p tmp/info
-       $(_SINGLE)$(NO_TRACE_MAKE) -j1 -r -s -f include/scan.mk SCAN_TARGET="packageinfo" SCAN_DIR="package" SCAN_NAME="package" SCAN_DEPTH=5 SCAN_EXTRA=""
-       $(_SINGLE)$(NO_TRACE_MAKE) -j1 -r -s -f include/scan.mk SCAN_TARGET="targetinfo" SCAN_DIR="target/linux" SCAN_NAME="target" SCAN_DEPTH=2 SCAN_EXTRA="" SCAN_MAKEOPTS="TARGET_BUILD=1"
-       for type in package target; do \
-               f=tmp/.$${type}info; t=tmp/.config-$${type}.in; \
-               [ "$$t" -nt "$$f" ] || ./scripts/$${type}-metadata.pl $(_ignore) config "$$f" > "$$t" || { rm -f "$$t"; echo "Failed to build $$t"; false; break; }; \
-       done
-       [ tmp/.config-feeds.in -nt tmp/.packageauxvars ] || ./scripts/feeds feed_config > tmp/.config-feeds.in
-       ./scripts/package-metadata.pl mk tmp/.packageinfo > tmp/.packagedeps || { rm -f tmp/.packagedeps; false; }
-       ./scripts/package-metadata.pl pkgaux tmp/.packageinfo > tmp/.packageauxvars || { rm -f tmp/.packageauxvars; false; }
+       #$(_SINGLE)$(NO_TRACE_MAKE) -j1 -r -s -f include/scan.mk SCAN_TARGET="packageinfo" SCAN_DIR="package" SCAN_NAME="package" SCAN_DEPTH=5 SCAN_EXTRA=""
+       #$(_SINGLE)$(NO_TRACE_MAKE) -j1 -r -s -f include/scan.mk SCAN_TARGET="targetinfo" SCAN_DIR="target/linux" SCAN_NAME="target" SCAN_DEPTH=2 SCAN_EXTRA="" SCAN_MAKEOPTS="TARGET_BUILD=1"
+       #for type in package target; do \
+       #       f=tmp/.$${type}info; t=tmp/.config-$${type}.in; \
+       #       [ "$$t" -nt "$$f" ] || ./scripts/$${type}-metadata.pl $(_ignore) config "$$f" > "$$t" || { rm -f "$$t"; echo "Failed to build $$t"; false; break; }; \
+       #done
+       #[ tmp/.config-feeds.in -nt tmp/.packageauxvars ] || ./scripts/feeds feed_config > tmp/.config-feeds.in
+       #./scripts/package-metadata.pl mk tmp/.packageinfo > tmp/.packagedeps || { rm -f tmp/.packagedeps; false; }
+       #./scripts/package-metadata.pl pkgaux tmp/.packageinfo > tmp/.packageauxvars || { rm -f tmp/.packageauxvars; false; }
        touch $(TOPDIR)/tmp/.build
```

See here:
```
lemoer@orange ~/d/f/g/g/openwrt> time make package/gluon-status-page/compile
 make[1] package/gluon-status-page/compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/gluon-web host-compile
 make[2] -C package/utils/lua host-compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/packages/lang/luasrcdiet host-compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/gluon-status-page compile
4.34user 0.71system 0:04.86elapsed 104%CPU (0avgtext+0avgdata 118076maxresident)k
0inputs+1080outputs (0major+235348minor)pagefaults 0swaps

```

lemoer, 2020-01-11:
Even before any target is printed, it still takes ~3 sec for preprocessing the Makefiles.
```
lemoer@orange ~/d/f/g/g/openwrt> date; time make package/gluon-status-page/compile -p | grep -e "Finished Make" -e make\\\[
Sat 11 Jan 2020 11:28:23 PM CET
# Finished Make data base on Sat Jan 11 22:28:25 2020
# Finished Make data base on Sat Jan 11 22:28:25 2020
# Finished Make data base on Sat Jan 11 22:28:25 2020
 make[1] package/gluon-status-page/compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/gluon-web host-compile
 make[2] -C package/utils/lua host-compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/packages/packages/lang/luasrcdiet host-compile
 make[2] -C /home/lemoer/documents/ff/git/gluon/package/gluon-status-page compile
# Finished Make data base on Sat Jan 11 23:28:28 2020
4.48user 0.90system 0:05.18elapsed 104%CPU (0avgtext+0avgdata 118228maxresident)k
0inputs+1080outputs (0major+234624minor)pagefaults 0swaps
```

NeoRaider, 2020-01-14:
- As soon as you switch to the `openwrt` directory, you aren't using any Gluon scripts anymore. We don't want to make any downstream modifications to the OpenWrt build system. The openwrt-devel ML would be a better place to talk about this
- During development, I rarely build packages at all. We could add a few scripts for quick deployments of Lua / Shell / any non-compiled code to a running device, possibly even with a watcher process for automatic deployment on save.
- You can always run a single build step using the commands displayed during build without any dependency handling, you just need to pass TOPDIR pointing to the OpenWrt toplevel dir. For example from `openwrt`:
```sh
make TOPDIR=$(pwd) -C ../package/gluon-status-page compile
```

NeoRaider, 2020-05-10:
Closing, as workarounds exist, and any further work should be discussed on the OpenWrt side.

closed 2020-05-10 by NeoRaider