freifunk-gluon/gluon #1923: Tunneldigger tunnel loses link to batman mesh

RalfJung, 2020-01-26:
### Bug report

**What is the problem?**

We have now at least twice on one of our nodes seen the following effect: the node drops offline (according to hopglass) even though it still has internet connectivity.  Doing `batctl gwl` shows only one of our 4 gateways, even though those are all directly connected to each other:
```
[B.A.T.M.A.N. adv openwrt-2018.1-9, MainIF/MAC: primary0/22:a2:aa:c0:65:63 (bat0/30:b5:c2:6e:9d:6b BATMAN_IV)]
  Router            ( TQ) Next Hop          [outgoingIf]  Bandwidth
* ca:fe:ba:be:00:03 ( 37) ca:fe:ba:be:00:03 [  mesh-vpn]: 250.0/250.0 MBit
```
A `batctl ping` to the gateway shown here does not work, but the TQ stays fixed (30 in one case, 37 in the other). `batctl n` shows
```
[B.A.T.M.A.N. adv openwrt-2018.1-9, MainIF/MAC: primary0/22:a2:aa:c0:65:63 (bat0/30:b5:c2:6e:9d:6b BATMAN_IV)]
IF             Neighbor              last-seen
     mesh-vpn     ca:fe:ba:be:00:03    5.720s
```

Restarting the tunneldigger broker on the gateway side fixes the issue. In one case, we have logs of when the issue started happening:

```
Sun Jan 26 14:16:13 2020 daemon.warn td-client: Tunnel has timed out, closing down interface.
Sun Jan 26 14:16:19 2020 daemon.err td-client: Connection to gw2.saar.freifunk.net lost.
Sun Jan 26 14:16:20 2020 daemon.info td-client: Performing broker selection...
Sun Jan 26 14:16:21 2020 daemon.debug td-client: Broker usage of gw1.saar.freifunk.net:10000: 23551
Sun Jan 26 14:16:21 2020 daemon.debug td-client: Broker usage of gw2.saar.freifunk.net:10000: 23551
Sun Jan 26 14:16:21 2020 daemon.debug td-client: Broker usage of gw4.saar.freifunk.net:10000: 23380
Sun Jan 26 14:16:21 2020 daemon.debug td-client: Broker usage of gw3.saar.freifunk.net:10000: 23380
Sun Jan 26 14:16:21 2020 daemon.info td-client: Selected gw3.saar.freifunk.net:10000 as the best broker.
Sun Jan 26 14:16:24 2020 daemon.info td-client: Tunnel successfully established.
```
Somehow, the then-established tunnel doesn't actually "work" though. But since batman mysteriously still thinks it sees one gateway, the watchdog(s) don't see any problem either, and the node just stays in limbo.

The first time this happened, quite a few nodes in the network dropped offline at the same point, so I think this affects many nodes, not just the one from which I got the above logs.

I don't have any good ideas for how to further debug this. What could make batman not realize that the connection is down, and leave some "residue" TQ? And why would the batman link have trouble in the first place, since the l2tp link looks fine?

**Gluon Version:** v2019.1.1

**Site Configuration:** https://git.hacksaar.de/FreifunkSaar/gluon-site

**Custom patches:** https://git.hacksaar.de/FreifunkSaar/gluon-ffsaar

rotanid, 2020-01-27:
don't you think this is more likely a problem in tunneldigger or batman-adv, not in Gluon?
if you agree, please open a bug report at these projects.

RalfJung, 2020-01-27:
I have no idea where to report or how to debug this, I'm afraid. This is a classical "integration" kind of issue, and I don't know how to even figure out which component is responsible. I figured in the gluon repo it's most likely that someone else with the same issue would see it.

mweinelt, 2020-01-27:
> What could make batman not realize that the connection is down, and leave some "residue" TQ?

Packet loss.

> And why would the batman link have trouble in the first place, since the l2tp link looks fine?

Does it? How did you measure that?

> Restarting the tunneldigger broker on the gateway side fixes the issue.

What exactly happens during such a restart?

T-X, 2020-01-27:
@RalfJung: Could you also post "batctl o" on the node? And ideally also "batctl o" and "batctl n" from the gateway.

If I'm rembering correctly, then "batctl n" was a uni-directional thing. So if the tunnel were defunct only in one direction, for instance from the node to the gateway then you'd still see updates in "batctl n" on the node.

Could you check whether "ping6" over the tunnel interface still works when "batctl ping" does not anymore?

RalfJung, 2020-01-31:
> Packet loss.

I don't follow. As far as I can tell, there is *no* connection between the node and the gateway. It's just dead. But still Batman claims a TQ of 37, but only for the "active"/"selected" gateway. The TQ should gradually go to 0, right? Is there some batman bug that means the selected gateway cannot drop to 0, or so?

> Does it? How did you measure that?

[the l2tp link]
Yeah, good point. I didn't see any error about it. But I am also not sure how to measure this.

> What exactly happens during such a restart?

The client tunneldigger gets a signal that the connection is going down, tears it down from its own end and re-initializes everything.

> Could you also post "batctl o" on the node? And ideally also "batctl o" and "batctl n" from the gateway.

I'll do that next time it happens.

> Could you check whether "ping6" over the tunnel interface still works when "batctl ping" does not anymore?

So, ping6 using the link-local `fe80` address of the `mesh-vpn`/`l2tp-xxx` interfaces (node/gateway) should work? That's good to know!

RalfJung, 2020-02-01:
> Could you check whether "ping6" over the tunnel interface still works when "batctl ping" does not anymore?

I tried this on a working node, and even there it doesn't work. According to the tunneldigger logs, right now l2tp245-245 is the interface connected to the node I have set up in a VM.
```
$ ip a s l2tp245-245
16951: l2tp245-245: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1406 qdisc pfifo_fast master saarVPN state UNKNOWN group default qlen 1000
    link/ether 8a:12:9d:a4:36:2d brd ff:ff:ff:ff:ff:ff
    inet6 fe80::8812:9dff:fea4:362d/64 scope link 
       valid_lft forever preferred_lft forever
```
So I tried `ping6 fe80::8812:9dff:fea4:362d%mesh-vpn` on the node, but there is no response. A `tcpdump -i l2tp245-245 icmp` on the gateway side doesn't show any packets. Even when I try a ping from the gateway to the node, the tcpdump on the gateway shows nothing and the ping says "Destination unreachable: Address unreachable" (which I suppose means it couldn't even determine the MAC address?).

I am not sure if there is really supposed to be working IPv6 in that tunnel, it looks to me like it's ethernet-only.

RalfJung, 2020-02-23:
I just had this happen again to a node I control. I wonder how often we lose other nodes due to this.

Is there any way to check the status of an established l2tp connection, when batman is acting up like here? I tried adding IPv4 addresses to both endpoints and pinging, and that did not work in the broken state, but I don't know if that means anything.

Adorfer, 2020-02-25:
> I tried adding IPv4 addresses to both endpoints and pinging, and that did not work in the broken state, but I don't know if that means anything.

won't help if there is a redundant link in the local cloud. 
i assume that you have to do some magic with screenscapring batctl o and look for the direct path between the uplink, memorize that and see if that (later) disappears for a prologued time.

RalfJung, 2020-02-25:
The node in question did not have any local cloud neighbors, only the defunct mesh-vpn link.
That's why I am so confused that anything at all shows up in `batctl gwl` -- the node is entirely separated from the mesh, but somehow batman doesn't realize that this is the case.

belzebub40k, 2020-02-25:
Maybe check `batctl n` instead of `batctl gwl` as it show when a neighbor was last seen. AFAIK it takes a moment until a neighbor is removed.

RalfJung, 2020-02-25:
I gave `batctl n` output in the OP -- it claims to have seen it but a few seconds ago. But at the same time, `batctl ping` to that supposed neighbor does not work.

T-X, 2020-02-25:
> I tried adding IPv4 addresses to both endpoints and pinging, and that did not work in the broken state, but I don't know if that means anything.

This sounds as if not batman-adv but the tunnel might be (partially?) broken.

@RalfJung: Could you try the following:

Checking multicast:

1) On node A, start a "ping6 -s 600 ff02::1%l2tp245-245"
2) On node A, check if you see the packets via a 'node-a$ tcpdump -i l2tp245-245 "greater 600 and less 1200"'
3) On node A, check if you see the packets on the underlying physical interface, for instance via  'node-a$ tcpdump -i eth0 "greater 600 and less 1200"'
4) On node B, check if you see the packets on the underlying physical interface, for instance via  'node-a$ tcpdump -i eth0 "greater 600 and less 1200"'
5) On node B, check if you see the packets via a 'node-a$ tcpdump -i l2tp245-245 "greater 600 and less 1200"'

Checking unicast:

Try the same as for multicast, but check if you see a packet of similar size, the ICMPv6 echo reply, in the opposite direction.

Also perform all the above with A and B reversed, that is running ping6 on node B. To check if unicast or multicast transmission might be broken in one direction only.

T-X, 2020-02-25:
If I remember correctly, then ``batctl n`` will update and reset the "last-seen" as long as it receives OGM packets (broadcast) from the other node. While "last-seen" in ``batctl o`` will only update if OGMs are both sent and received, I think.

So when checking ``batctl n`` best check if it updates on both sides.

Also, if ``batctl n`` will update if broadcasting on the tunnel is working but unicasting isn't. That'd best be checked via "ping6" and "tcpdump" as described above.

Also if you could post both ``batctl n`` and ``batctl o`` from both nodes could be helpful.

RalfJung, 2020-02-26:
>   This sounds as if not batman-adv but the tunnel might be (partially?) broken.

Well, yeah, the tunnel is certainly not fully working any more. But even then I'd hope batman would just cease to see *any* neighbors/gateways, which would make the watchdog trigger and restart the entire thing so that it all works again.

> Could you try the following:

Ah, so IPv6 multicast like that should work even if `l2tp245-245` only has an `fe80` address? Or should I assign some other address first?

> Also if you could post both batctl n and batctl o from both nodes could be helpful.

I'll gather that data the next time this happens.

T-X, 2020-02-27:
> Ah, so IPv6 multicast like that should work even if l2tp245-245 only has an fe80 address? Or should I assign some other address first?

Right, ``ff02::1`` is a link-local IPv6 multicast address. So an IPv6 link-local unicast address, fe80::..., is sufficient, the IPv6 stack will then pick that as the source address for the multicasted ICMPv6 packet ping6 generates.

> Well, yeah, the tunnel is certainly not fully working any more. But even then I'd hope batman would just cease to see any neighbors/gateways, which would make the watchdog trigger and restart the entire thing so that it all works again.

Hm, for ``batctl n`` and ``batctl o`` we added the ``-t <timeout>`` option. We could add the same thing for ``batctl gwl``. However that would still leave us with issues if only unicast and not broadcast/multicast is broken which batman-adv would not be able to detect via its OGMs. Or we wouldn't be able to detect MTU issues for instance either. Or odd-numbered packets laggy before noon ;-) (https://xkcd.com/2259/).

Maybe it would make sense to add some occasional ping6 tests or similar to the tunneldigger watchdog?

RalfJung, 2020-02-27:
> Or odd-numbered packets laggy before noon ;-) (https://xkcd.com/2259/).

Yeah, that XKCD was published very timely and made me laugh hard. ;)

> Hm, for batctl n and batctl o we added the -t <timeout> option

So that will only show neighbors that have been seen within the timeout? But the thing is, `batctl n` shows "seen in <10s", so I doubt it would help here.

> Maybe it would make sense to add some occasional ping6 tests or similar to the tunneldigger watchdog?

Yeah, that sounds like a good idea. And when using the IPv6 multicast "all node" address, we don't even have to know the IP address on the other side.

However, we'll have to be a bit careful... when I do this on a working node, I get duplicates:
```
# ping6 -s 600 ff02::1%mesh-vpn
PING ff02::1%mesh-vpn (ff02::1%47): 600 data bytes
608 bytes from fe80::20a2:aaff:fec0:6567: seq=0 ttl=64 time=0.592 ms
608 bytes from fe80::c8fe:baff:febe:4: seq=0 ttl=64 time=33.959 ms (DUP!)
608 bytes from fe80::20a2:aaff:fec0:6567: seq=1 ttl=64 time=0.563 ms
608 bytes from fe80::c8fe:baff:febe:4: seq=1 ttl=64 time=18.465 ms (DUP!)
608 bytes from fe80::20a2:aaff:fec0:6567: seq=2 ttl=64 time=0.676 ms
608 bytes from fe80::c8fe:baff:febe:4: seq=2 ttl=64 time=18.425 ms (DUP!)
608 bytes from fe80::20a2:aaff:fec0:6567: seq=3 ttl=64 time=0.560 ms
608 bytes from fe80::c8fe:baff:febe:4: seq=3 ttl=64 time=16.695 ms (DUP!)
608 bytes from fe80::20a2:aaff:fec0:6567: seq=4 ttl=64 time=0.609 ms
608 bytes from fe80::c8fe:baff:febe:4: seq=4 ttl=64 time=16.360 ms (DUP!)
```
So looks like we'll have to filter out responses from ourselves?

Adorfer, 2020-02-27:
> The node in question did not have any local cloud neighbors, only the defunct mesh-vpn link

that's what you described, yes. 
my comment referred to the watchdog which should trigger even if there is a second vpn link in the same local mesh cloud. 
For that you have to e.g. either ping6 the link-local opposite of the l2tp tunnel. or even to determine the batid of the rmote, batctl ping it and then look if the batctl o routing still is "direct".

rotanid, 2020-12-14:
@RalfJung is this still an open issue and have you been debuggen it further, maybe documented somewhere else?

RalfJung, 2020-12-14:
We recently had a node drop offline for an hour and then coming back on its own without having rebooted, but we have no idea what that might be caused by... it's really hard to debug these things. :/

I have not seen this specific issue in a while, but I also have not be "on the hunt" for it.

rotanid, 2020-12-14:
@RalfJung feel free to re-open the issue if you encounter it specifically and/or can contribute something new to it

closed 2020-12-14 by rotanid