freifunk-gluon/gluon #1936: Move specialised packages to the community repo?

CodeFetch, 2020-02-20:
Some packages would better fit in the community repository as they are not essential for Gluon.
This might make the Gluon core project more generic (as a mesh framework) and therefore lighten the maintenance burden for the Gluon developers by making it more attractive to people who don't want to run Freifunk with it.

Following come to my mind:
- gluon-check-connection
- gluon-offline-ssid
- gluon-scheduled-domain-switch 
- gluon-hoodselector
- maybe gluon-alt-esc
- maybe gluon-status-page

rotanid, 2020-02-20:
this is not what the community repository is meant for.
bluntly said, the repo is for low quality packages.

mweinelt, 2020-02-20:
I'm definately vetoing `gluon-scheduled-domain-switch` and `gluon-status-page`, they are too much at the core of Gluon.

rotanid, 2020-02-21:
... and as gluon-check-connection is planned as an advanced version of a check currently included in gluon-scheduled-domain-switch, this also needs good quality and maintainers as gatekeepers ;)

2tata, 2020-02-21:
I advise to hold gluon-hoodselector also closely because there will follow additional features witch makes the package way more closer to the framework due to the planed scan mode and others.

CodeFetch, 2020-02-21:
> bluntly said, the repo is for low quality packages.

Hm. "low quality" packages sounds like something that should not be offered at all. For me community packages are packages that go beyond the scope of what is ordinarily being used, but mainly packages by "communities" for "communities" and staging packages which don't have the quality for being used as a main package, yet. At first I thought like you that all packages should become part of the Gluon core project, but giving it a second thought I think the maintenance burden for the core developers and the barrier to entry  for newcomers is quite high and the packages look more like a restriction than being modular at first sight.

What is Gluon? `a modular framework for creating OpenWrt-based firmwares for wireless mesh nodes` like the description says or `a framework for creating OpenWrt-based community mesh network firmwares`.

As I think the Gluon core should be a generic wireless mesh node firmware building kit, things like the hoodselector and the status-page are packages relevant for "community networks" and thus fit better in the community repository. Otherwise I'd propose to rename the community package repository to staging-packages or something.

blocktrron, 2020-02-21:
> Hm. "low quality" packages sounds like something that should not be offered at all. For me community packages are packages that go beyond the scope of what is ordinarily being used, but mainly [...]

The goal for the community packages is to enable packages to have shorter iteration periods, thus they can be merged quicker into a central repository, usable for all communities. The goal of the community repository is not to move existing packages there.

This is the reason why I'm closing this issue now. Feel free to open a pull-request in the community repository to improve it's description.

> As I think the Gluon core should be a generic wireless mesh node firmware building kit, things like the hoodselector and the status-page are packages relevant for "community networks" and thus fit better in the community repository.

Gluon is already a generic mesh-framework. e.g. People use it for non-freifunk projects in the nordics. I'll give you the point Gluon has constraints in what network topologies are supported, however there is currently no one pushing actively for such an idea, nor with a roadmap how to achieve this generalization. 

> My initial idea was to have stable and testing branches in the community repository for distinguishing between "low quality" and "mature" packages. My intention was not a rubbish-repository.

It is not planned to be a rubbish repository. The statement from @rotanid might be a bit misleading, but again: Out goal is to provide a place for Gluon features which are provided from communities for communities without the long review-cycles seen in the main project. This also means, we do not have to take care in the main projects of all packages incompatibility. This is (hence the name) the job of the package maintainers and the community as a whole.

On a sidenote: The "Move packages in the hands of the community" worked out great for LEDE / OpenWrt.

closed 2020-02-21 by blocktrron

SvenRoederer, 2020-02-29:
>  People use it for non-freifunk projects in the nordics

@blocktrron do you have a reference?