freifunk-gluon/gluon #1937: Potential breakage of Ubiquiti EdgeRouter-X with v2020.1 and above

blocktrron, 2020-02-25:
### Bug report

**What is the problem?**
Existing Ubiquiti EdgeRouter-X with Gluon 2019.x or earlier carry a high risk of ending up in a soft-bricked state when updating to a Gluon release based on OpenWrt 19.07 (and later).

This is due to the fact of the MT7621 NAND driver shifting the mtd layer on bad blocks instead of providing a 1:1 flash map in Versions up to OpenWrt 18.06. When upgrading to a OpenWrt 19.07 based Gluon release, the UBI blockmap becomes inconsistent, as the NAND driver now behaves as expected.

The breakage of an update can be determined beforehand - the NAND driver prints information about Bad blocks to the kernel ringbuffer.

```
[    1.184858] [NAND]select ecc bit:4, sparesize :64 spare_per_sector=16
[    1.197724] nand: device found, Manufacturer ID: 0xc2, Chip ID: 0xda
[    1.210365] nand: Macronix NAND 256MiB 3,3V 8-bit
[    1.219723] nand: 256 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
[    1.234822] Scanning device for bad blocks
[    1.528783] Bad eraseblock 1167 at 0x0000091e0000
[    1.696618] Bad eraseblock 1815 at 0x00000e2e0000
[    1.762803] 6 fixed-partitions partitions found on MTD device MT7621-NAND
[    1.776314] Creating 6 MTD partitions on „MT7621-NAND“:
[    1.786719] 0x000000000000-0x000000080000 : „u-boot“
[    1.797690] 0x000000080000-0x0000000e0000 : „u-boot-env“
[    1.809287] 0x0000000e0000-0x000000140000 : „factory“
[    1.820353] 0x000000140000-0x000000440000 : „kernel1“
[    1.831467] 0x000000440000-0x000000740000 : „kernel2“
[    1.842541] 0x000000740000-0x00000ff00000 : „ubi“
[    1.854888] [mtk_nand] probe successfully!
```

Log courtesy of @oszilloskop 

This device has two bad blocks, which will render the device bricked when attempting to upgrade.

The device can be recovered by re-flashing the device using the factory instructions with an OpenWrt initramfs image of version 19.07 or higher.

**Gluon Version:**
Gluon 2020.1 and above

**Site Configuration:**
Not applicable

**Custom patches:**
Not applicable

txt-file, 2020-02-28:
IMHO this should be reported upstream. I found no report on [bugs.openwrt.org](bugs.openwrt.org).
Is it normal that ubnt ER-X have corrupt flash? I expect every device with broken/corrupt flash to have problems to update.

oszilloskop, 2020-02-28:
There is a bug report, but only regarding the symptom.
The kernel log because of bad blocks and the result (kernel panic) are identical with my ER-X.
https://bugs.openwrt.org/index.php?do=details&task_id=2713

blocktrron, 2020-02-28:
@txt-file 

The bug was with the old behavior, which is fixed now. The breaking sysupgrade is a side-effect.

Broken blocks from factory are "normal" with NAND flash, but the bad block handling should be done on flash-translation layer, not the driver. This management is not necessary with NOR flash.

Note that the OpenWrt sysupgrade procedure is fundamentally different for NAND flash.

PR #1958 (2020-03-19): docs: v2020.1: add Edgerouter X NAND driver problem to known issues
In 9e4eb182908320cb94974663c7d201ce792eea78 I missed adding information wrt to #1937, that should be fixed with this PR.

As the issue is not actionable to us I will close it after this gets merged and backported.

Resolves #1937
commit bf041070 (2020-03-19): docs: v2020.1: add Edgerouter X NAND driver problem to known issues

Resolves #1937
commit 7669aea6 (2020-03-19): docs: v2020.1: add Edgerouter X NAND driver problem to known issues

Resolves #1937

(cherry picked from commit bf04107048d68cfb9f0aab19ba76b4a53fea92b5)
closed 2020-03-19 by blocktrron