freifunk-gluon/gluon #1939: the nodes time should be displayed or exported

Noodlesalat, 2020-02-29:
We are currently migrating our nodes into the multidomain network. The nodes are migrating if a certain date is reached.
While the migration we also discovered that on some devices the local date is somewhere in the past (e.g. Aug 2019) and as a result these nodes do not migrate. We are going to investigate this (maybe NTP-) issue, but I do not think that it is gluon related at this point.

Maybe the local (nodes) time should displayed or exported somewhere (e.g. status page, respondd) in order to prevent such time related issues. What do you think?

mweinelt, 2020-04-04:
The node time does not seem like a sensible value to expose. It would probably make more sense to expose metrics from the NTP daemon, although that may prove difficult to impossible with the busybox implementation.

cuechan, 2020-05-30:
We recently had a similar issue where nearly all nodes just started updating instantly despite the `PRIORITY=7` in the manifest. We think that there was a problem with our NTP server. A way to check if a node has the correct time would be very helpful.

NeoRaider, 2020-05-30:
I just intended to add a "time_synchronized" attribute to the respondd "statistics" object (the `adjtimex()` function can be used to query synchronization status from the kernel), but it turned out that we already export the current time since #1287. NTP sync status could be added as well, but seems somewhat redundant to me now. @mweinelt

closed 2020-05-30 by NeoRaider

rotanid, 2020-05-30:
anyone feel free to add the already exported time to the status page ;-)