freifunk-gluon/gluon #1940: AVM FRITZ!WLAN Repeater 450E doesn't reboot after flashing from v2018.2.4 to 2020.1

Byggvir, 2020-02-29:
I flashed two AVM FRITZ!WLAN Repeater 450E from v2018.2.4 to 2020.1. 
The repeater hangs during reboot with a fast flashing Power LED.

blocktrron, 2020-02-29:
Fast flashing Power-LED sounds like OpenWrt failsafe (which should be disabled in Gluon.

Does the device itself work with a current OpenWrt 19.07 build?

Byggvir, 2020-03-01:
Didn't try OpenWrt 19.07 on the device.

Byggvir, 2020-03-01:
Connected to the Repeater on 192.168.1.1 The device is in Failsafe Mode.

blocktrron, 2020-03-01:
Can you be more specific on the Failsafe mode? Is this the behavior you see on a OpenWrt 19.07 or a Gluon build (or both)?

Additionally, could you paste the output of `cat /sys/kernel/debug/gpio` with no buttons pressed?

blocktrron, 2020-03-01:
Can you try a build from the latest master branch? It should solve your issue

The active-low flag seems to be incorrect on this board.

```
root@xxxxxxxxxxx:~# cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 0-31, parent: platform/ath79-gpio, ath79-gpio:
 gpio-4   (                    |WPS Button          ) in  lo    
 gpio-11  (                    |PHY reset           ) out hi    
 gpio-12  (                    |MDC Pull-UP         ) out hi    
 gpio-13  (                    |fritz450e:green:lan ) out hi    
 gpio-14  (                    |fritz450e:green:powe) out hi    
 gpio-15  (                    |fritz450e:green:wlan) out lo    
 gpio-16  (                    |fritz450e:green:rssi) out hi    
 gpio-17  (                    |fritz450e:green:rssi) out hi    
 gpio-18  (                    |fritz450e:green:rssi) out hi    
 gpio-19  (                    |MDIO Pull-UP        ) out hi    

gpiochip1: GPIOs 488-511, ath9k-phy0:
 gpio-489 (                    |ath9k-phy0          ) in  lo
```

(Tanks to @Noodlesalat )

Byggvir, 2020-03-01:
It's the behavior of a Gluon build  v2020.1.
/sys/kernel/degbug is empty.

```
BusyBox v1.30.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 19.07-SNAPSHOT, r10915+8-aed6632d31
 -----------------------------------------------------
================= FAILSAFE MODE active ================
special commands:
* firstboot          reset settings to factory defaults
* mount_root     mount root-partition with config files

after mount_root:
* passwd                         change root's password
* /etc/config               directory with config files

for more help see:
https://openwrt.org/docs/guide-user/troubleshooting/
- failsafe_and_factory_reset
- root_password_reset
=======================================================

root@(none):~# cat /sys/kernel/debug/gpio
cat: can't open '/sys/kernel/debug/gpio': No such file or directory
root@(none):~# 
```

blocktrron, 2020-03-01:
I've had wrong memories - Gluon only disables the rest button handling while keeping the failsafe fully functional.

Does the latest master sow the same misbehavior? I assume https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=1d4f2ca6105698fde3b997c32e7e41f080c79a15 fixes the issue, which is already in master. If it fixes the issue for you, i will include the change in the v2020.1.x branch, so it can be fixed with Gluon 2020.1.1

Byggvir, 2020-03-02:
I create an image from Gluon master and just flashed the router from v2018.2.4 to master. 
With Gluon master the repeater works fine as I can see after a few minutes.
See: http://[2a03:2260:3017:1400:f2b0:14ff:fe2b:56fa]/cgi-bin/status

blocktrron, 2020-03-02:
The fix is now also in v2020.1.x and will be included in the upcoming service release.

rotanid, 2020-03-07:
@blocktrron can we close this issue? usually we don't keep them open until there's a release?

mweinelt, 2020-03-07:
Fixed in v2020.1.1.

closed 2020-03-07 by mweinelt