freifunk-gluon/gluon #1941: TP-Link Archer C7 v2: Both WLAN LED are always off

oszilloskop, 2020-03-01:
### Bug report

**What is the problem?**
With Gluon v2020.1 a TP-Link Archer C7 v2 does not activate the 2,4GHz-WLAN-LED and the 5GHz-WLAN-LED anymore.

**What is the expected behaviour?**
Both TP-Link Archer C7 v2 WLAN LED are blinking as usual.

**Gluon Version:**
Gluon v2020.1

**Site Configuration:**
https://github.com/freifunk-ffm/site-ffffm/tree/stable

**Custom patches:**
none

blocktrron, 2020-03-01:
Should be fixed by https://github.com/freifunk-gluon/gluon/commit/add560b18f1db642bd19dbc4e3ef95c3a334d535 on master. Can you confirm this?

Will also be picked up on v2020.1.x on the next bump.

oszilloskop, 2020-03-02:
> Should be fixed by [add560b](https://github.com/freifunk-gluon/gluon/commit/add560b18f1db642bd19dbc4e3ef95c3a334d535) on master. 

I can confirm that add560b fixed the Issue.

blocktrron, 2020-03-02:
Okay, i Will pick this to v2020.1.x later.

I will keep this issue open as soon as Gluon 2020.1.1 is released, so it stays visible.

blocktrron, 2020-03-02:
The fix is now also in v2020.1.x and will be included in the upcoming service release.

mweinelt, 2020-03-07:
Fixed in v2020.1.1.

closed 2020-03-07 by mweinelt