freifunk-gluon/gluon #1942: Missing release notes of: Removing UCI manipulation of 'hop_penalty'

oszilloskop, 2020-03-02:
The feature to manipulate the value of the Batman mesh parameter `hop_penalty` via UCI was removed in Gluon v2010.1, but it is not mentioned in the release notes.

This feature was a common way to change the value of `hop_penalty` and save it persistently. It is well documented in many Freifunk wikis.  Communitys need a hint to adjust their wikis.

The reason why I go with the problem this way, and do not open a PR is, I don't know much about the background and nothing about the documentation process of Gluon.
 
Gluon Version:
v2020.1

blocktrron, 2020-03-02:
The hop_penalty was never intended to be user nor site-selectable. Even before switching to Gluon's netifd mesh-proto, the hop_penalty was overwritten on each invocation of `gluon-reconfigure` or upgrade.

However, customization could be enabled, by adding a community package. This package could be hooked to the mesh proto and allow custom configuration of batman's parameters. But this is just a quick idea from my side.

Adorfer, 2020-03-02:
just for note: this uci was used on many communities in local meshes with multiple uplinks with large bandwith unequality. e.g. when one "10x slower" leg was going into heavy packet loss under load, causing oscillation routes. 
The "non-persistancy" of the setting was normally adjusted by putting a check in /etc/rc.local to reinstate the value after updates.

blocktrron, 2020-03-02:
@Adorfer Setting the hop_penalty was never mentioned in Gluons documentation. The forced overwrite in Gluons upgrade scripts should have made it clear, that there was (at no given point) support in Gluon for this configuration option.

This Issue could be closed here, however (as you and @oszilloskop seem to be interested in this feature), I wanted to point out a feasible path on how to implement this (and maybe more) feature.

oszilloskop, 2020-03-03:
If it is a problem to mention the removal of the formerly used batman-adv UCI sections (e.g. hop_penalty) in the release notes, then this issue can be closed at this point. 

OT:
The possibility to modify the hop penalty is a must have for me. I have to use the `/etc/rc.local` in the future. Hooking a community package to netifd mesh-proto is too much dangerous because to many unknown Gluon changes could happen in the future.

blocktrron, 2020-03-03:
I disagree with your stand of it being to dangerous, as it can be implemented in a way it only interfaces with the routing protocol. Nevertheless, a PR implementing it in the main project is also welcome.

rotanid, 2020-03-03:
> The "non-persistancy" of the setting was normally adjusted by putting a check in /etc/rc.local to reinstate the value after updates.

using rc.local to change it still works, the exact way changed a bit, someone mentioned it on the forum i think.

by the way, i even didn't know about this setting before...

closed 2020-04-20 by NeoRaider

NeoRaider, 2020-04-20:
Oops, I pushed that to master by accident instead of creating a new branch for a PR... see the commit message for the implemented UCI settings.

commit 778bf905 (2020-04-20): gluon-mesh-batman-adv: add UCI setting for hop penalty

Add a UCI setting gluon.mesh_batman_adv.hop_penalty

Example UCI commands:

    uci set gluon.mesh_batman_adv=mesh_batman_adv
    uci set gluon.mesh_batman_adv.hop_penalty=20
    uci commit

`/etc/config/gluon` config section:

    config mesh_batman_adv 'mesh_batman_adv'
    	option hop_penalty '20'

Fixes: #1942