freifunk-gluon/gluon #1943: Fritzbox 7330 ethernet connection only when connected to LAN1 before power up

rabbit83ka, 2020-03-02:
To be able to connect to the Router (Config-Mode / normal mode) the ethernet cable has to be plugged in LAN1 upon booting. If it's not connected at the moment of powering up the device, no connection is possible (even the ports of a switch don't light up then). LAN2 doesn't work at all. Same behaviour in Gluon and OpenWRT.

**What is the expected behaviour?**

Connection should be established when you plug in the ethernet cable, no matter when.

**Gluon Version:**

v2020.1

**Site Configuration:**

https://github.com/ffka/site-ffka/blob/master/site.conf

**LInks**

Same Bug in [OpenWRT](https://bugs.openwrt.org/index.php?do=details&task_id=2240&string=fritz&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)

blocktrron, 2020-03-03:
Ping @Dark4MD

blocktrron, 2020-04-08:
I've spoke to @Dark4MD on IRC. Obviously this device has major flaws hence why it should not have been added to Gluon in the first place. He can however confirm this issue on his unit.

Please try the attached patch with OpenWrt master. If it at least fixes the issue with LAN1 we can discuss about the device not being removed from Gluon.

```
From 20c20635fd9b6c5b49229cb48468b0e10e553f40 Mon Sep 17 00:00:00 2001
From: David Bauer <mail@david-bauer.net>
Date: Wed, 8 Apr 2020 14:49:39 +0200
Subject: [PATCH] lantiq: add AVM FRITZ!Box 7320 LAN1 PHY

The AVM FRITZ!Box 7320 LAN1 port currently only works when the port is
up on boot. Otherwise the bootloader asserts the PHY reset.

Also fix the obviously incorrect PHY mode, as the Lantiq PEF7071 only
support RGMII.

---
 .../mips/boot/dts/lantiq/ar9_avm_fritz7320.dts    | 15 ++++++++++++++-
 .../mips/boot/dts/lantiq/ar9_avm_fritz7320.dts    | 15 ++++++++++++++-
 2 files changed, 28 insertions(+), 2 deletions(-)

diff --git a/target/linux/lantiq/files-4.19/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts b/target/linux/lantiq/files-4.19/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
index 424b778bb4..01cc986710 100644
--- a/target/linux/lantiq/files-4.19/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
+++ b/target/linux/lantiq/files-4.19/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
@@ -74,9 +74,22 @@
 };
 
 &gsw {
-	phy-mode = "mii";
+	phy-mode = "rgmii";
 	mtd-mac-address = <&ath9k_cal 0xa91>;
 	mtd-mac-address-increment = <(-2)>;
+
+	phy-handle = <&phy0>;
+
+	mdio-bus {
+		#address-cells = <1>;
+		#size-cells = <0>;
+
+		phy0: ethernet-phy@0 {
+			compatible = "lantiq,phy11g", "ethernet-phy-ieee802.3-c22";
+			reg = <0>;
+			reset-gpios = <&gpio 34 GPIO_ACTIVE_LOW>;
+		};
+	};
 };
 
 &localbus {
diff --git a/target/linux/lantiq/files-5.4/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts b/target/linux/lantiq/files-5.4/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
index 424b778bb4..01cc986710 100644
--- a/target/linux/lantiq/files-5.4/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
+++ b/target/linux/lantiq/files-5.4/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
@@ -74,9 +74,22 @@
 };
 
 &gsw {
-	phy-mode = "mii";
+	phy-mode = "rgmii";
 	mtd-mac-address = <&ath9k_cal 0xa91>;
 	mtd-mac-address-increment = <(-2)>;
+
+	phy-handle = <&phy0>;
+
+	mdio-bus {
+		#address-cells = <1>;
+		#size-cells = <0>;
+
+		phy0: ethernet-phy@0 {
+			compatible = "lantiq,phy11g", "ethernet-phy-ieee802.3-c22";
+			reg = <0>;
+			reset-gpios = <&gpio 34 GPIO_ACTIVE_LOW>;
+		};
+	};
 };
 
 &localbus {
-- 
2.25.1
```

Ping @Dark4MD

closed 2020-04-26 by blocktrron

rabbit83ka, 2020-05-03:
I've tested the patch above. The Box (Fritzbox 7330) now answers to pings and is reachable via ssh after unplugging and replugging the ethernet. So LAN1 is working as expected. Thanks. Hope it will be back/stay in gluon.

blocktrron, 2020-05-03:
So the device is now reachable via LAN1 when booted without a cable plugged and connected when the device finished booting?

rabbit83ka, 2020-05-03:
Sorry, to early. I did some further testing and LAN1 still doesn't work when unplugged at boot...

everloop2, 2020-05-04:
> I've spoke to @Dark4MD on IRC. Obviously this device has major flaws hence why it should not have been added to Gluon in the first place. He can however confirm this issue on his unit.
> 
> Please try the attached patch with OpenWrt master. If it at least fixes the issue with LAN1 we can discuss about the device not being removed from Gluon.
> 
> ```
> From 20c20635fd9b6c5b49229cb48468b0e10e553f40 Mon Sep 17 00:00:00 2001
> From: David Bauer <mail@david-bauer.net>
> Date: Wed, 8 Apr 2020 14:49:39 +0200
> Subject: [PATCH] lantiq: add AVM FRITZ!Box 7320 LAN1 PHY
> 
> The AVM FRITZ!Box 7320 LAN1 port currently only works when the port is
> up on boot. Otherwise the bootloader asserts the PHY reset.
> 
> Also fix the obviously incorrect PHY mode, as the Lantiq PEF7071 only
> support RGMII.
> 
> Signed-off-by: David Bauer <mail@david-bauer.net>
> ---
>  .../mips/boot/dts/lantiq/ar9_avm_fritz7320.dts    | 15 ++++++++++++++-
>  .../mips/boot/dts/lantiq/ar9_avm_fritz7320.dts    | 15 ++++++++++++++-
>  2 files changed, 28 insertions(+), 2 deletions(-)
> 
> diff --git a/target/linux/lantiq/files-4.19/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts b/target/linux/lantiq/files-4.19/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
> index 424b778bb4..01cc986710 100644
> --- a/target/linux/lantiq/files-4.19/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
> +++ b/target/linux/lantiq/files-4.19/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
> @@ -74,9 +74,22 @@
>  };
>  
>  &gsw {
> -	phy-mode = "mii";
> +	phy-mode = "rgmii";
>  	mtd-mac-address = <&ath9k_cal 0xa91>;
>  	mtd-mac-address-increment = <(-2)>;
> +
> +	phy-handle = <&phy0>;
> +
> +	mdio-bus {
> +		#address-cells = <1>;
> +		#size-cells = <0>;
> +
> +		phy0: ethernet-phy@0 {
> +			compatible = "lantiq,phy11g", "ethernet-phy-ieee802.3-c22";
> +			reg = <0>;
> +			reset-gpios = <&gpio 34 GPIO_ACTIVE_LOW>;
> +		};
> +	};
>  };
>  
>  &localbus {
> diff --git a/target/linux/lantiq/files-5.4/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts b/target/linux/lantiq/files-5.4/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
> index 424b778bb4..01cc986710 100644
> --- a/target/linux/lantiq/files-5.4/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
> +++ b/target/linux/lantiq/files-5.4/arch/mips/boot/dts/lantiq/ar9_avm_fritz7320.dts
> @@ -74,9 +74,22 @@
>  };
>  
>  &gsw {
> -	phy-mode = "mii";
> +	phy-mode = "rgmii";
>  	mtd-mac-address = <&ath9k_cal 0xa91>;
>  	mtd-mac-address-increment = <(-2)>;
> +
> +	phy-handle = <&phy0>;
> +
> +	mdio-bus {
> +		#address-cells = <1>;
> +		#size-cells = <0>;
> +
> +		phy0: ethernet-phy@0 {
> +			compatible = "lantiq,phy11g", "ethernet-phy-ieee802.3-c22";
> +			reg = <0>;
> +			reset-gpios = <&gpio 34 GPIO_ACTIVE_LOW>;
> +		};
> +	};
>  };
>  
>  &localbus {
> -- 
> 2.25.1
> ```
> 

did test build kernel 4.14 - LAN1 still bugged

everloop2, 2020-05-04:
FritzBox 7320 and FritzBox7330 have [different network controllers](https://bugs.openwrt.org/index.php?do=details&task_id=2240&string=fritz&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=).

mweinelt, 2020-05-05:
Upstream-Bug https://bugs.openwrt.org/index.php?do=details&task_id=2240

blocktrron, 2020-05-05:
I assume @everloop2 came from the upstream bug ;)

As there are major flaws, this case is (for now) closed in regards to Gluon. When both boards are fixed properly upstream, re-adding them might be possible but for now they won't return.

PR #1995 (2020-04-26): lantiq-xway: remove support for AVM FRITZ!Box 7320 7330 7330SL
This device has broken Ethernet on both ports.

Remove support for those devices. for now, as there was no feedback from
the original author.

Closes #1943
commit 5231fb01 (2020-04-26): lantiq-xway: remove support for AVM FRITZ!Box 7320 7330 7330SL

This device has broken Ethernet on both ports.

Remove support for those devices. for now, as there was no feedback from
the original author.

Closes #1943
commit 81d6b9c1 (2020-04-29): lantiq-xway: remove support for AVM FRITZ!Box 7320 7330 7330SL

This device has broken Ethernet on both ports.

Remove support for those devices. for now, as there was no feedback from
the original author.

Closes #1943

(cherry picked from commit 5231fb017800975bddaf655fa988055e007884be)