freifunk-gluon/gluon #1944: Potential OOM in ath9k-based (and maybe other) devices

CodeFetch, 2020-03-03:
### Bug report

Since Gluon v2017.1 we've seen low memory and OOM (out-of-memory) situations on a range of devices (most of them with 32 MB RAM).
All of them were based on the ath9k chipset. A memory leak happens when a reference to allocated memory gets lost and thus the memory never gets freed. This is not the case here. 
Instead it seems that either a buffer without proper size limitation is being filled occasionally or the devices are just OOM, in which case we would expect the OOM killer to reboot the device as a last resort, which does not or only rarely happen. Instead we see a high load occuring which has been tracked down to be caused by the decompression of flash memory of the squashfs file system in low-memory-situations. This is a thrashing issue which means that pages are constantly being freed and reallocated due to the low memory situation. Ordinarily the thrashing issue should be detected, but it is a [known problem](https://www.spinics.net/lists/kernel/msg2653186.html) of squashfs and is unlikely to see a fix as such a low memory situation should not happen at all.

Devices which were upgraded to double the amount of RAM don't show these problems unless they have a vast amount of WLAN clients. It seems that [somewhere](https://github.com/freifunk-gluon/gluon/issues/1768#issuecomment-515795087) in the WLAN stack less skbs are being freed than are being allocated. After some time or in low-memory-situations they still get freed, but too late. This is a hint to stale skbs in WLAN packet reassembly as the rejects in frame reassembly are expected to be much higher in crowded environments with a lot of WLAN clients.

This might also be related to #1880.

Assuming the issue exists in the packet reassembly routine, for finding out about what exactly happens, it is needed to [patch the SLUB](https://github.com/freifunk-gluon/gluon/commit/b6b61d6999af3d853d1789e4eeac5d6689477e5b) debugging procedure as the dumping otherwise slows the device down too much and leads to a reboot and desirably adding more sophisticated tracing routines. Furthermore the issue must be triggered in a controlled environment as there is only a fine line between the bug occuring and unresponsiveness. This could e.g. be achieved by simulating a lot of WLAN clients using some ath9k-based WLAN routers and creating [multiple STA interfaces](https://github.com/freifunk-gluon/gluon/issues/1768#issuecomment-506368366).

tackin, 2020-03-03:
this is: #1768

Adorfer, 2020-10-23:
i am seeing OOM now on gluon-v2020.2.1-5-g1c73b043+  (kernel  4.14.195)

`kern.warn kernel: [  941.299984] swapper: page allocation failure: order:0, mode:0x1080020(GFP_ATOMIC), nodemask=(null)`

full logs: https://termbin.com/92qb 

hypothesis: related to https://github.com/freifunk-gluon/gluon/commit/9891e01d835f66436df8262c483e53251327fed2

since it does not happen on  gluon-v2020.1.3-8-g3ef69c96+ which is kernel 4.14.180

blocktrron, 2020-10-23:
Have you tried removing zram-swap from the package set?

CodeFetch, 2020-10-23:
This looks like the kernel CONFIG_SWAP is not selected... As zram can be used without a swap, it is not selected by default from what I can see. I wonder why it was selected before... I don't know how to fix it atm...

Adorfer, 2020-10-24:
> I don't know how to fix it atm...

full buildlogs can be provided if needed. 
anything i can check for? or to activate zram?
Where is the kernal configfile to check on the build system corresponding to this?
https://www.acmesystems.it/ts_swapon_failed

CodeFetch, 2020-10-25:
@Adorfer Sorry very much, but I misunderstood the option. It is not a kernel config parameter, but a busybox config parameter and from what I can see it is set.

Adorfer, 2020-10-25:
just for my understanding

1) how to avoid that? What to reconfigure to avoid this "swapon: Function noimplemented" 
2) what is the consequence? Swap not turned on, 13MB RAM laying unused?
3) does it work on devices where it should work by default (like ubiquiti-nanostation-m according to https://github.com/freifunk-gluon/gluon/blob/v2020.2.x/targets/ar71xx-generic

blocktrron, 2020-10-26:
@Adorfer ar71xx-tiny has swap support disabled by defaul (I haven't paid attention to your log and were under the impression, this was on ar71xx-generic).

So you'll need to turn on the CONFIG_KERNEL_SWAP symbol in OpenWrt in order to use swap. This symbol is enabled by default on all targets which do not have the SMALL_FLASH flag assigned to them (which ar71xx-tiny does)

Otherwise you are right, you are wasting quite a lot of RAM without and benefit in return.

As explained above, ar71xx-generic targets such as nanostations should not be affected by this and work as expected.

Adorfer, 2020-10-26:
> So you'll need to turn on the CONFIG_KERNEL_SWAP symbol in OpenWrt in order to use swap

where is this set?

> i tried to add the Kernel options in openwrt,
> https://github.com/eulenfunk/firmware/blob/85ab5b20216604dba4ff5f03b669f51f6f916b1d/patches/zram-ar71xx-tiny-kernel.patch

as stated above this was not effective.

blocktrron, 2020-10-26:
You are editing the target kernel configuration, while the SWAP symbol there is overwritten by OpenWrt's buildsystem.

You can try this (not tested):

```
--- a/scripts/target_config_lib.lua
+++ b/scripts/target_config_lib.lua
@@ -158,6 +158,8 @@ if not lib.opkg then
        lib.packages {'-opkg'}
 end

+lib.config('KERNEL_SWAP', true)
+
 if #lib.devices > 0 then
        handle_target_pkgs(lib.target_packages)
```

CodeFetch, 2022-08-08:
I close it as such issues have not been observed for a while and don't appear with sufficient memory.

CodeFetch, 2022-08-08:
Test

closed 2022-08-08 by CodeFetch

Adorfer, 2022-08-10:
We keep seeing this OOM happening on devices with 64M as well when surpassing around 120 assigned clients. 
But since there is no advance in this topic, this stays WontFix.  (specially since those devices were never designed for such situations. recommendation: if you have to many clients, reduce your txpower drastically)