freifunk-gluon/gluon #1952: tunneldigger-watchdog often says "Process-Pid does not match with pid-File."

RalfJung, 2020-03-08:
### Bug report

I noticed that I am seeing the tunneldigger-watchdog kick in with "Process-Pid does not match with pid-File" much more often that I would expect, and even when tunneldigger and the tunnel are in perfect health as far as I can see.

**What is the problem?**
tunneldigger-watchdog restarts tunneldigger even when there is no reason to do so, leading to unnecessary brief connectivity loss.

**What is the expected behaviour?**
tunneldigger-watchdog should not unnecessarily restart tunneldigger.

**Problem analysis:**
Looking at the watchdog, it seems to determine the current tunneldigger PID by looking at the first line of `pgrep tunneldigger`. However, that will show *all* processes called tunneldigger. Another such process is `tunneldigger-watchdog` itself. My guess for what happens is that whenever the watchdog happens to have a PID lower than tunneldigger itself, it sees a wrong PID and then restarts tunneldigger.

I think the watchdog should switch to `pgrep -f "^/usr/bin/tunneldigger "` to make sure that it only finds processes with that binary, not other processes with "tunneldigger" in their name.

**Gluon Version:** v2019.1.2

**Site Configuration:** https://git.hacksaar.de/FreifunkSaar/gluon-site/-/blob/master/site.conf

**Custom patches:** https://git.hacksaar.de/FreifunkSaar/gluon-ffsaar

mweinelt, 2020-03-08:
This change sounds sensible enough, can you create a pull request?

RalfJung, 2020-03-08:
Done: https://github.com/freifunk-gluon/gluon/pull/1953

closed 2020-03-08 by mweinelt