freifunk-gluon/gluon #1961: config-mode, autoupdater: Names of branches are ignored

citronalco, 2020-03-25:
According to the gluon docs it is possible to give the different firmware branches names, like in this site.conf excerpt:

```
autoupdater = {
    branch = 'stable',
    branches = {
        stable = {
            name = 'Maximum stability',
            [...]
        },
        experimental = {
            name = 'Bleeding edge',
            [...]
```
I would expect that the "name" would show up in Luci/Config Mode's Autoupdater module, so that  the users could choose between "Maximum stability" and "Bleeding edge". But the web interface shows only "stable" and "experimental" - what's in "name" seems to be competely irrelevant.

Additionally the order in which the branches are shown in the web interface seems to be pretty random and neither in the same order as configured in site.conf (what I would expect) nor alphabetically.

This affects latest Gluon (2020.1), 2019 and probably all olders, too.

blocktrron, 2020-03-25:
The list is not sorted, as the site generates uci entries which in return are used to generate the dropdown in the web view.

For using the name setting in the web-view you can try this patch:

```
From cd2fbfce9a0dd29516ac8c006b39c3d81f864632 Mon Sep 17 00:00:00 2001
From: David Bauer <mail@david-bauer.net>
Date: Wed, 25 Mar 2020 17:31:43 +0100
Subject: [PATCH] gluon-web-autoupdater: use name key for prettyname

---
 .../luasrc/lib/gluon/config-mode/model/admin/autoupdater.lua    | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/package/gluon-web-autoupdater/luasrc/lib/gluon/config-mode/model/admin/autoupdater.lua b/package/gluon-web-autoupdater/luasrc/lib/gluon/config-mode/model/admin/autoupdater.lua
index adf7cbd6..7cd1e13e 100644
--- a/package/gluon-web-autoupdater/luasrc/lib/gluon/config-mode/model/admin/autoupdater.lua
+++ b/package/gluon-web-autoupdater/luasrc/lib/gluon/config-mode/model/admin/autoupdater.lua
@@ -24,7 +24,7 @@ end
 o = s:option(ListValue, "branch", translate("Branch"))
 uci:foreach("autoupdater", "branch",
 	function (section)
-		o:value(section[".name"])
+		o:value(section[".name"], section["name"])
 	end
 )
 o.default = uci:get("autoupdater", autoupdater, "branch")
-- 
2.26.0
```

citronalco, 2020-03-25:
Thanks for the patch, I'll try!

Simply omitting "name" breaks compilation:
```
[...]
Configuring kmod-ipt-offload.
Configuring urngd.
Configuring gluon-config-mode-mesh-vpn.
Setting flags for package libc to hold.
Setting flags for package libgcc1 to hold.
Setting flags for package libpthread to hold.
Setting flags for package librt to hold.
cp -fpR /home/citronalco/freifunk/gluon/openwrt/build_dir/target-mips_24kc_musl/root-ar71xx /home/citronalco/freifunk/gluon/openwrt/build_dir/target-mips_24kc_musl/root.orig-ar71xx
*** site.conf error: expected autoupdater.branches.experimental.name to be a string, but it is unset
postinst script ./usr/lib/opkg/info/gluon-autoupdater.postinst has failed with exit code 1
make[3]: *** [package/Makefile:71: package/install] Error 1
make[3]: Leaving directory '/home/citronalco/freifunk/gluon/openwrt'
make[2]: *** [package/Makefile:108: /home/citronalco/freifunk/gluon/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_install] Error 2
make[2]: Leaving directory '/home/citronalco/freifunk/gluon/openwrt'
make[1]: *** [/home/citronalco/freifunk/gluon/openwrt/include/toplevel.mk:227: world] Error 2
make[1]: Leaving directory '/home/citronalco/freifunk/gluon/openwrt'
make: *** [Makefile:157: all] Error 2
```

NeoRaider, 2020-03-30:
With the current site.conf structure, the best we could do is sort by label - which isn't too bad I guess.

PR #1984 (2020-04-26): gluon-web-autoupdater: use human-readable names as branch labels, sort by label
Use the value of the `name` site.conf field as label (it was
accidentally unused before).

Our site.conf currently doesn't define a specific order for the branch
entries. To avoid changing branch orders, sort entries by this label.

Fixes: #1961
commit 696b4316 (2020-04-26): gluon-web-autoupdater: use human-readable names as branch labels, sort by label

Use the value of the `name` site.conf field as label (it was
accidentally unused before).

Our site.conf currently doesn't define a specific order for the branch
entries. To avoid changing branch orders, sort entries by this label.

Fixes: #1961
closed 2020-04-26 by mweinelt