freifunk-gluon/gluon #1969: remove comments in firmware images

rubo77, 2020-04-05:
There are 925 lines of comments in an unpacked firmware image, which are about 42KB (13kb zipped) which could be removed:

https://gist.github.com/rubo77/74ee2e045d3e1e1b7fce0c9ad1099a0a

I used this to detect:

```
binwalk -e gluon-firmwareXXX.bin
cd _gluon-firmwareXXX.bin.extracted/squashfs-root
rm var # which is a symlink to tmp
grep '^\s*#[^!]' -Rh .|egrep -v '(Binär)'|cat>../comments_in_gluon.txt 
$ wc ../comments_in_gluon.txt
  981  6810 42798 ../comments_in_gluon.txt

NeoRaider, 2020-04-05:
- The README files exist to ensure that these directories are always created (an empty .keep file could be used instead, but it doesn't really matter)
- posix.lua is part of openwrt/packages -> not a Gluon issue
- simple-uci.lua is in freifunk-gluon/packages and shoudn't depend on Gluon's GluonSrcDiet macro; adding another config symbol seems like overkill for a few comments in a single file
- I assume most of these comments are in Shell scripts. Stripping them must be done in OpenWrt, which has been rejected before; therefore, we will reject it for Gluon as well (13KB just isn't worth maintaining downstream patches forever)

closed 2020-04-05 by NeoRaider

rubo77, 2020-04-07:
Couldn't we add a shell minifyer before we create the images, that removes all comments (and maybe gain some more kb by miniifying other stuff) in the system before it creates the firmware image?

2tata, 2020-04-11:
On 07/04/2020 09:50, Ruben Barkow-Kuder wrote:
> Couldn't we add a shell minifyer before we create the images, that removes all comments (and maybe gain some more kb by miniifying other stuff) in the system before it creates the firmware image?

That was what I have tried to merge into OpenWRT some time ago. I think
that it what @NeoRaider refers above.
But I think it was manly refuse because of the use of Haskell ... I
don't remember the discussion well.

LG
Tarek

rubo77, 2020-04-12:
@2tata can you link to your pr?