freifunk-gluon/gluon #1985: uci set batman-adv.bat0.hop_penalty

4ndr3, 2020-04-19:
Bitte den Wert batman-adv.bat0.hop_penalty wieder zu uci hinzufuegen.

NeoRaider, 2020-04-19:
Hmm, in the new UCI structure, we should put it somewhere in `/etc/config/gluon` (`/etc/config/batman-adv` does not exist anymore). I propose to use one of the following options:

- `gluon.mesh.hop_penalty`
- `gluon.mesh_batman_adv.hop_penalty`
- `gluon.mesh.batman_adv_hop_penalty`

blocktrron, 2020-04-19:
@NeoRaider 

How about `gluon.batman_adv.hop_penalty`? Mesh seems redundant to me (at least in this context).

mweinelt, 2020-04-19:
Duplicate of #1942.

NeoRaider, 2020-04-19:
@blocktrron I think we should align the section names with package names, and the package is called gluon-mesh-batman-adv(-15).

rotanid, 2020-04-19:
closing, as @mweinelt mentioned this is a duplicate if #1942 , please continue over there, thank you!

closed 2020-04-19 by rotanid