freifunk-gluon/gluon #1987: RFC: Offical way supporting additional Mesh- and Client-Interfaces as "user"-defined-interface

margau, 2020-04-21:
Hello together,

there are use cases, where a gluon-node needs more than "just" WAN, Client, or the respective Mesh on WAN/Mesh on LAN-Interfaces. 

For example, I sometimes have setups using three Interfaces - WAN, Client and Mesh - or even more. Another use case would be a "BATMAN-Router" between more than two RF-Links. 

As far as I know, there is no official way, but some unofficial (legacy) manuals like https://wiki.freifunk-stuttgart.net/anleitungen:mesh_und_switche.

I'd love to see the functionality in Gluon Upstream.
My "dream" would be the possibility to configure additional interfaces/VLAN's reboot save and officially supported via UCI, at a later step maybe via config mode, for each necessary "type" (Mesh, Client, WAN, Private...).

I'm suggesting something like the following configuration:
```
config interface 'user_lan'
	option ifname 'eth2'
	option proto ('gluon_user_wired'|'gluon_user_client'|'gluon_user_wan')
	option disabled '0'
```

What are the thoughts about this? Is there already work done, or where are potential caveats?

This issue is similar to #1670 and a few other, older issues.

rotanid, 2020-04-22:
don't take this as a NACK, but i have some offload-setups where i also set up WAN/Client/Mesh and i only needed to set up either Client or Mesh manually. Also, this is possible in a  mostly update-safe way, at least i didn't have problems so far. You need to add a custom section to /etc/config/network which Gluon won't touch. In some cases, you also may want to change the value of /lib/gluon/core/sysconfig/lan_ifname or /lib/gluon/core/sysconfig/wan_ifname
I reccomend to don't change WAN if possible, to have additional security that the node at least comes online after an update, so the worst case would be to set the custom config up again remotely. (But for most updates, even this is not necessary)

margau, 2020-06-20:
An update on this issue: The mesh-part is solved by #2041, I've proposed documentation in #2059. 

At the moment still not clear is how to handle the interface-index properly, as it is limited to 3 bits and all 8 MAC's are already used (see https://github.com/freifunk-gluon/gluon/blob/7fdffcaa1c3cf1b8f5591385d6aa18c15f038f7a/package/gluon-core/luasrc/usr/lib/lua/gluon/util.lua#L139).

Adorfer, 2020-06-23:
this will hopefully end the days of ugly rc.local hack jobs which currently prevent nodes from losing their setup/function after autoupdates.

maurerle, 2023-08-28:
This issue can be closed, as a way to configure such interfaces is implemented in #2688.
Official Documentation of how to change this is on the cli is given here: https://gluon.readthedocs.io/en/latest/features/wired-mesh.html#commandline

closed 2026-01-03 by maurerle