freifunk-gluon/gluon #1988: ATH10K and 64MB RAM

jenell95, 2020-04-22:
### Bug report

**What is the problem?**
sporadically OOM and kernel panics on devices with 64 MB RAM and ATH10K

**What is the expected behaviour?**
As many of us know the ATH10K driver is not resource friendly at all. So the OpenWRT forum says dualband devices should have at least 128 MB RAM. Unfortunately the patch for 64 MB RAM has been removed in upstream and won't support mesh. But in single band usage it works fine.

So my idea was to have it run in either 2,4 GHz or 5 GHz. This can be done by a simple change of the checkboxes/radiobuttons in luci. If 2.4 GHz is marked the 5 GHz will be unmarked and vice versa. 

**Gluon Version:**
maybe all 

**Site Configuration:**
maybe all

**Custom patches:**
none

blocktrron, 2020-04-22:
Affected devices will show spurious OOM  even in medium load environments with only the ath10k radio active. So they are only really usable when used 2.4 GHz only.

belzebub40k, 2020-04-22:
Even if the 5GHz radio is disabled by default or only one of them is usable I would still consider these devices broken. I as a user who bought such a device would expect that both radios are work in parallel like with the stock firmware.

jenell95, 2020-04-22:
The device is only available in aftermarkets. I don't think a DAU would buy this. And a DAU couldn't flash the firmware. 

I think people who buy these will be ok with it. I will.

blocktrron, 2020-04-26:
Closing this, as the reason why there isn't a non-hacky solution to our problem.

If you want to push this forward upstream (in the form of a ath10k-smallbuffers package for example) we can incorporate this in Gluon.

closed 2020-04-26 by blocktrron