freifunk-gluon/gluon #1993: RFC: docs: Checklist for device integration

AiyionPrime, 2020-04-25:
Thanks to several community members and our docs, I just added my first profile for a new router.
The process was quite straight forward, as far as the technical aspects were concerned.

One thing missing though, was a checklist [like this one](https://github.com/freifunk-gluon/gluon/pull/1986#issue-406621144) in the docs for the PR.
The PR is often a hassle on its own, with a lot of small improvement commits over a relative long time.
I understand, quality of content is necessary and hopefully the inclusion of a checklist in the docs helps to reduce the ping-pong-factor further.

I hope, the checklist will also include things like hints regarding expected formatting in the docs.

rotanid, 2020-04-28:
hm, isn't this what you are looking for?
https://github.com/freifunk-gluon/gluon/wiki/Device-Integration-checklist

AiyionPrime, 2020-04-28:
Yes, indeed, a great start. If this was part of the documentation (or at least referenced?), and not well hidden in githubs wiki, it might be more helpful.

This list could be accompanied with the most common suggestions for improvements of PRs regarding new devices.
Like hints about the formatting, how to finalize the PR by rebasing it into one feature commit and so on.

The goal would be more profile-PRs, that don't need to be refactored over the course of a few days.

2tata, 2020-05-01:
It also took me some time to find this list again ;P

PR #2003 (2020-05-02): docs: dev: add link to device checklist
Closes #1993
commit daf8a6c1 (2020-05-02): docs: dev: add link to device checklist

Closes #1993
closed 2020-05-02 by mweinelt