freifunk-gluon/gluon #2009: Kernel panic in 2020.1.2 (generic x86)

ChristophFranzen, 2020-05-05:
Hallo zusammen, ein [Gluon 2020.1.2 (x86; Freifunk Trier)](https://github.com/freifunktrier/firmware_store/blob/master/firmware/stable/factory/gluon-fftr-0.13.4%2Btackin-x86-64.img.gz) fällt hier auf die Nase. Es haut sich weg in dem Augenblick, wo der Kernel init starten will:
```
traps: init[1] trap invalid opcode ip:b7f39508 sp:bfdafb30 error:0 in libc.so[b7f05000+4b000]
Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004
```
Der Prozessor ist ein P-III, ältere Linux-Kernel (Jessie, ein altes Gluon v2016.1.6-3-g871460b, M0n0wall/pfSense [BSD]) gehen auf dem Gerät. Das war neu aufgesetzt, weil Sysupgrade von dem alten Gluon nicht ging.
### Hardware:
[NORCO_MB-X67.pdf](https://github.com/freifunk-gluon/gluon/files/4580999/2015.07.16.22.52.01.NORCO_MB-X67.pdf)
> Intel Pentium III/Celeron Processor in Socket 370 (133 MHz), support B-stepping
> Intel 815EP chipset, IT8712F I/O Chipset
> Two Intel 82540 Ethernet controller Four Intel 82551, One on-chip MAC + Intel 82562
> One 168pin DIMM Socket, support PC-100/133 MHz SDRAM
> One 40-pin (2x20, 2mm) IDE connector that support Ultra ATA100, build in secondary master IDE Channel
> One 44-pin (2x22, 2mm) IDE connector that support Ultra ATA100, build in secondary slave IDE Channel
> One CompactFlash type II socket, support ATA mode, build in primary master IDE Channel
> AMI licensed BIOS Seven RJ-45 external connectors One RS-232 DB-9 connector
### Ganzes Log (die paar fehlenden Zeilen hat es mir vor dem Kopieren überschrieben, die seltsamen Zeichen kommen von der seriellen Kosole):
```
[    0.000000] Linux version 4.14.176 (tackin@debian) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r10915+8-aed6632d31)) #0 SMP Wed Feb 5 19:10:43 2020
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000fffe0000-0x00000000ffffffff] reserved
[    0.000000] Notice: NX (Execute Disable) protection missing in CPU!
[    0.000000] Legacy DMI 2.3 present.
[    0.000000] DMI: American Megatrends Inc.         Uknown                          /Unknown                         , BIOS 07.00T                
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] e820: last_pfn = 0x20000 max_arch_pfn = 0x1000000
[    0.000000] x86/PAT: PAT not supported by CPU.
[    0.000000] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC  
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI BIOS Error (bug): A valid RSDP was not found (20170728/tbxfroot-244)
[    0.000000] 0MB HIGHMEM available.
[    0.000000] 512MB LOWMEM available.
[    0.000000]   mapped low ram: 0 - 20000000
[    0.000000]   low ram: 0 - 20000000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.000000]   Normal   [mem 0x0000000001000000-0x000000001fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.000000]   node   0: [mem 0x0000000000100000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000001000-0x000000001fffffff]
[    0.000000] Using APIC driver default
[    0.000000] Local APIC disabled by BIOS -- you can enable it with "lapic"
[    0.000000] APIC: disable apic facility
[    0.000000] APIC: switched to apic NOOP
[    0.000000] smpboot: Allowing 1 CPUs, 0 hotplug CPUs
[    0.000000] e820: [mem 0x20000000-0xfffdffff] available for PCI devices
[    0.000000] Booting paravirtualized kernel on bare hardware
[    0.000000] clocksource: refined-jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.000000] random: get_random_bytes called from start_kernel+0x6c/0x3aa with crng_init=0
[    0.000000] setup_percpu: NR_CPUS:4 nr_cpumask_bits:4 nr_cpu_ids:1 nr_node_ids:1
[    0.000000] percpu: Embedded 28 pages/cpu s81932 r0 d32756 u114688
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129950
[    0.000000] Kernel command line: BOOT_IMAGE=/boot/vmlinuz failsafe=true root=PARTUUID=ffd22553-02 rootfstype=squashfs rootwait console=tty0 console=ttyS0,115200n8 noinitrd libata.force=pio4
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Initializing CPU#0
[    0.000000] Initializing HighMem for node 0 (00000000:00000000)
[    0.000000] Memory: 508168K/523896K available (7084K kernel code, 330K rwdata, 2408K rodata, 648K init, 364K bss, 15728K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] virtual kernel memory layout:
[    0.000000]     fixmap  : 0xfff67000 - 0xfffff000   ( 608 kB)
[    0.000000]   cpu_entry : 0xffa00000 - 0xffa9d000   ( 628 kB)
[... 138 lines cut ...]
[    1.616138] NET: Registered protocol family 10
[    1.625702] Segment Routing with IPv6
[    1.629462] NET: Registered protocol family 17
[    1.633967] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    1.636988] 8021q: 802.1Q VLAN Support v1.8
[    1.642343] mce: Using 5 MCE banks
[    1.646050] microcode: sig=0x6b4, pf=0x10, revision=0x0
[    1.651339] microcode: Microcode Update Driver: v2.2.
[    1.651347] Using IPI No-Shortcut mode
[    1.660215] sched_clock: Marking stable (1656433804, 0)->(1655621891, 811913)
[    1.667640] registered taskstats version 1
[    1.672985] rtc_cmos rtc_cmos: setting system clock to 2003-10-29 13:05:45 UTC (1067432745)
[    1.681678] clocksource: tsc: mask: 0xffffffffffffffff max_cycles: 0x11561343af0, max_idle_ns: 440795243708 ns
[    1.692059] clocksource: Switched to clocksource tsc
�[    1.950790] VFS: Mounted root (squashfs filesystem) readonly on device 8:2.
^@[    1.960597] Freeing unused kernel memory: 648K
[    1.985530] Write protecting the kernel text: 7088k
[    1.990665] Write protecting the kernel read-only data: 2436k
^@[    2.002663] random: fast init done
[    2.160751] traps: init[1] trap invalid opcode ip:b7f39508 sp:bfdafb30 error:0 in libc.so[b7f05000+4b000]
^@[    2.170594] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004
[    2.170594] 
[    2.179841] Kernel Offset: disabled
[    2.183337] Rebooting in 1 seconds..
```

kpanic23, 2020-05-05:
Hm, looks like the init binary tries to use an opcode the Pentium III doesn't understand.
IIRC, OpenWrt switched from i686 to i386_pentium4 with the shift to LEDE in 2017.
You might need to build a x86-legacy image.
OTAH, it might be about time to replace the system with hardware from this century... ;)

ChristophFranzen, 2020-05-05:
My dear kpanic23, I can't tell you how much I dislike remarks like "hardware from this century", this would certainly be considered to be rude. It has become far too common to address minor technical issues by suggesting to throw away this old crap and buy something new. This is NOT what the world needs, at least when you don't deny climate change to be human-made.

csamsel, 2020-05-05:
Operating a system that has a power draw per instruction which is like hundred times higher then what's on the market isnt climate friendly.

ChristophFranzen, 2020-05-05:
Unless you take into consideration how much power is consumed by manufacturing a new device. How long must a power-hungry machine run to "outrun" a newly-built machine?

csamsel, 2020-05-05:
depends on the case obviously. I work in cloud computing, there hardware gets replaced in less then 3 years because after that the utlity costs outweight the investment in new hardware.
For embedded devices that's probably longer.

My rule of thumb efficiency-wise (i did some simple math) is that everything older then DDR3 (introduced 2007) should not be turned on at all.

ChristophFranzen, 2020-05-05:
In economy, it's all about "costs" – which means money – and since you don't necessarily pay more for greater pollution or energy consumption, this isn't really what we need to consider here.

csamsel, 2020-05-05:
Yes - cloud computing is an extreme example anyways.

What i wanted to say, don't think you do the climate any good by using a system from around 2000 (or how old it actually is), There might be other good reasons though.

edeso, 2020-05-05:
On 05.05.2020 18:51, Christian Samsel wrote:
> Yes - cloud computing is an extreme example anyways.
>
> What i wanted to say, don't think you do the climate any good by using a system from around 2000 (or how old it actually is), There might be other good reasons though.
>

guys. this is about gluon here. please move your off-topic issue somewhere else. thanks.. ede

ChristophFranzen, 2020-05-05:
The main reasons are that it still works, it's fast enough for the purpose, and it's already here. The climate aspect isn't easy to calculate, but generally: buying something new means to spend money now, having the same issue in a few years, it's brought from china by ships consuming the dirtiest fuel ever, the factories over there probably aren' at current standards as well…

ChristophFranzen, 2020-05-05:
Yes, about Gluon, @edeso, and I think you cannot make decisions on what to (still) support and what not only by "guessing" a fixed number of X years of existence.

blocktrron, 2020-05-05:
@kpanic23 pointed out what is required to make the system boot on those ancient legacy systems.

Discussions about whether or not to operate those inefficient hardware can be done [elsewhere](https://forum.freifunk.net/), as Issues are not the proper place to have those.

ChristophFranzen, 2020-05-06:
Auf Anfrage von @txt-file hier die Augabe von „lspci -knn“:
```
00:00.0 Host bridge [0600]: Intel Corporation 82815 815 Chipset Host Bridge and Memory Controller Hub [8086:1130] (rev 04)
 Subsystem: Intel Corporation 82815 815 Chipset Host Bridge and Memory Controller Hub [8086:1130]
 Kernel driver in use: agpgart-intel
00:01.0 PCI bridge [0604]: Intel Corporation 82815 815 Chipset AGP Bridge [8086:1131] (rev 04)
00:1e.0 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev 05)
00:1f.0 ISA bridge [0601]: Intel Corporation 82801BA ISA Bridge (LPC) [8086:2440] (rev 05)
 Kernel driver in use: lpc_ich
00:1f.1 IDE interface [0101]: Intel Corporation 82801BA IDE U100 Controller [8086:244b] (rev 05)
 Subsystem: Intel Corporation Device [8086:2440]
 Kernel driver in use: ata_piix
02:06.0 Unassigned class [ff00]: SafeNet Inc SafeXcel 1141 [16ae:1141] (rev 01)
 Subsystem: SafeNet (wrong ID) SafeXcel 1141 v. 1.1 [0001:0001]
02:09.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139] (rev 20)
 Subsystem: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139]
 Kernel driver in use: 8139cp
02:0a.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139] (rev 20)
 Subsystem: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139]
 Kernel driver in use: 8139cp
02:0b.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139] (rev 20)
 Subsystem: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139]
 Kernel driver in use: 8139cp
02:0c.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139] (rev 20)
 Subsystem: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139]
 Kernel driver in use: 8139cp
02:0d.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139] (rev 20)
 Subsystem: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139]
 Kernel driver in use: 8139cp
02:0e.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139] (rev 20)
 Subsystem: Realtek Semiconductor Co., Ltd. RTL-8100/8101L/8139 PCI Fast Ethernet Adapter [10ec:8139]
 Kernel driver in use: 8139cp
```

ChristophFranzen, 2020-05-06:
Anmerkungen zur PCI-Hardware:
1. 16ae:1141: „SafeXcel 1141“: Crypto-Mini-PCI-Karte, die wahrscheinlich weder einen freien, noch einen entgeltfrei erhältlichen Treiber hat. In der Original-Firewall-Software wird das benutzt.
2. 8086:244b: Intel 82801BA IDE, kann maximal PIO4, kein DMA, der Kernel sollte daher mit „_libata.force=pio4_“ gestartet werden. Das Modul „_ata-piix_“ ist UPSTREAM nicht im Geode-Zweig enthalten (der ansonsten bislang auf dieser Hardware zu laufen scheint), wohl aber in _Legacy_ und _Generic_.
3. 10ec:8139: RTL8139C+ ist anhand der ID leider nicht von älteren 81xx zu unterscheiden, braucht aber ZWINGEND an Stelle des _8139too_ das Modul _8139cp_, das in Openwrt (x86) standardmäßig nicht enthalten ist, in Gluon inzwischen schon, siehe hierzu: [Unterstützung für Netzwerkkarte Realtek 8139C+ in x86-generic](https://github.com/freifunk-gluon/gluon/issues/1022)

txt-file, 2020-05-14:
@ChristophFranzen Does setting "libata.force=pio4" survive sysupgrade?

txt-file, 2020-05-16:
I tried x86-legacy (https://github.com/freifunk-gluon/gluon/pull/2010) some hours ago on a WatchGuard FireBox x1250e. During boot it had some DMA issues as the hardware only supports PIO4 instead of DMA. The libata kernel driver in gluon master issued 3 soft-resets of the ata-port and changed first from to MWDMA2, then to MWDMA1 then to PIO4 and then continued to boot. Each soft reset took ~30 seconds. Because of these soft-resets I did not need the "libata.force=pio4" boot parameter. I did not test if x86-generic works on this machine.

An Asus eeePC 701 4G bootet also without issues with x86-legacy. ([Intel Dothan-512](https://en.wikipedia.org/wiki/List_of_Intel_Celeron_microprocessors#%22Dothan-512%22_(90_nm)) basierte CPU). The netbook did also boot with x86-generic.

closed 2020-05-17 by NeoRaider

ChristophFranzen, 2020-06-13:
Am Thu, 14 May 2020 04:36:31 -0700 schrieb txt-file
<notifications@github.com>:
> @ChristophFranzen Does setting "libata.force=pio4" survive sysupgrade?
 
Short answer: untested, and not strictly necessary.

Mich haben vom Test bisher 2 Dinge abgehalten:

1. Dieser Rechner war abgeraucht (Root-Dateisystem), einen Tag später
die letzte Vollsicherung und gestern auch noch mein Monitor. Inzwischen
ist alles halbwegs gerettet…

2. Wegen Git, das ganz offenbar IMMER Merge-Konflikte produziert, wenn
man was länger nicht anfaßt, an dem andere weiterfrickeln, konnte ich
nicht „mal eben“ nebenher eine Version bauen, die sowohl „ata_piix“, als
auch „8139cp“ enthält; wenn Du ein Image zur Hand hast, das „Legacy“
mit diesen beiden enthält, kann ich mal schnell testen, sonst dauert
das vielleicht noch was.

Grüße, Christoph

ChristophFranzen, 2020-06-13:
Hello,

Am Sat, 16 May 2020 16:38:15 -0700 schrieb txt-file
<notifications@github.com>:
> I tried x86-legacy
> (https://github.com/freifunk-gluon/gluon/pull/2010) some hours ago on
> a WatchGuard FireBox x1250e. During boot it had some DMA issues as
> the hardware only supports PIO4 instead of [...]

the e-series is a bit more recent, but similar. Indeed the X-Core
machine boots without the kernel parameter, but rather slowly dur to
those resets.

Christoph

PR #2010 (2020-05-17): RFC: add x86-legacy subtarget
As x86-generic is compiled to pentium4 (and newer) there is a need for
a subtarget for older devices. The x86-legacy subtarget is set to
compile to pentium (and newer) and should support even very old devics.

Comments welcome.

Closes: #2009