freifunk-gluon/gluon #2012: Gluon 2019.1.2 on TP-Link CPE 210 V3.0 - Bootloop

F1ndus, 2020-05-06:
### Bug report

**What is the problem?**
 
- CPE-210 V3.0 Bootloops forever on Gluon 2019.1.2, every version above works fine.

**What is the expected behaviour?**

- It should boot

**Gluon Version:**

https://github.com/freifunk-gluon/gluon/commit/63ebeb25c0b3a7eec4c70efc92dfcfa760dd85b6

**Site Configuration:**

https://gitli.stratum0.org/ffbs/ffbs-site/-/tree/master
0dc42bfe544e37d8c0da7aa7029ba4e198c7afd2

I am trying to get gluon 2019.1.2 to work on a TP-Link CPE-210 V3.0, but at least on this version the router gets stuck at booting gluon and loops endlessly.

2020.1.1 and above are working flawlessly, but the Freifunk-Community here is currently limited to BATMAN_IV_Legacy, so I have to stick with 2019.1.2 at the moment.

I tried to do a git bisect to find the commit where it starts working again (v2019.1.2 to v2020.1.1), some changes between dc8d5d3c and ba341505 fixes it, but applying a patch of these changes to 2019.1.2 did not help.

Patch from the differences between dc8d5d3c and ba341505 :

```diff
diff --git a/docs/user/supported_devices.rst b/docs/user/supported_devices.rst
index 5515dc9a..354ec7b7 100644
--- a/docs/user/supported_devices.rst
+++ b/docs/user/supported_devices.rst
@@ -13,10 +13,6 @@ ar71xx-generic
   - AP121 [#deprecated]_
   - AP121F
   - AP121U [#deprecated]_
-  - Hornet-UB [#deprecated]_
-  - Tube2H [#deprecated]_
-  - N2 [#deprecated]_
-  - N5 [#deprecated]_
 
 * Allnet
 
diff --git a/targets/ar71xx-generic b/targets/ar71xx-generic
index 92812b43..161ca795 100644
--- a/targets/ar71xx-generic
+++ b/targets/ar71xx-generic
@@ -4,10 +4,10 @@ config 'CONFIG_TARGET_SQUASHFS_BLOCK_SIZE=64'
 local ATH10K_PACKAGES = {}
 local ATH10K_PACKAGES_QCA9887 = {}
 local ATH10K_PACKAGES_QCA9888 = {}
-if env.GLUON_WLAN_MESH == 'ibss' then
-	ATH10K_PACKAGES = {'-kmod-ath10k', 'kmod-ath10k-ct', '-ath10k-firmware-qca988x', 'ath10k-firmware-qca988x-ct'}
-	ATH10K_PACKAGES_QCA9887 = {'-kmod-ath10k', 'kmod-ath10k-ct', '-ath10k-firmware-qca9887', 'ath10k-firmware-qca9887-ct'}
-	ATH10K_PACKAGES_QCA9888 = {'-kmod-ath10k', 'kmod-ath10k-ct', '-ath10k-firmware-qca9888', 'ath10k-firmware-qca9888-ct'}
+if env.GLUON_WLAN_MESH == '11s' then
+	ATH10K_PACKAGES = {'kmod-ath10k', '-kmod-ath10k-ct', 'ath10k-firmware-qca988x', '-ath10k-firmware-qca988x-ct'}
+	ATH10K_PACKAGES_QCA9887 = {'kmod-ath10k', '-kmod-ath10k-ct', 'ath10k-firmware-qca9887', '-ath10k-firmware-qca9887-ct'}
+	ATH10K_PACKAGES_QCA9888 = {'kmod-ath10k', '-kmod-ath10k-ct', 'ath10k-firmware-qca9888', '-ath10k-firmware-qca9888-ct'}
 end
 
 
@@ -24,21 +24,6 @@ device('alfa-network-ap121f', 'ap121f', {
 	factory = false,
 })
 
-device('alfa-network-hornet-ub', 'hornet-ub', {
-	profile = 'HORNETUB',
-	aliases = { 'alfa-network-ap121', 'alfa-network-ap121u' },
-	deprecated = true,  -- kernel partition too small with OpenWrt 19.07
-})
-
-device('alfa-network-tube2h', 'tube2h-8M', {
-	profile = 'TUBE2H8M',
-	deprecated = true,  -- kernel partition too small with OpenWrt 19.07
-})
-device('alfa-network-n2-n5', 'alfa-nx', {
-	profile = 'ALFANX',
-	deprecated = true,  -- kernel partition too small with OpenWrt 19.07
-})
-
 
 -- Allnet
 
diff --git a/targets/ar71xx-nand b/targets/ar71xx-nand
index 3ba2671a..dcedccb8 100644
--- a/targets/ar71xx-nand
+++ b/targets/ar71xx-nand
@@ -1,8 +1,8 @@
 config 'CONFIG_GLUON_SPECIALIZE_KERNEL=y'
 
 local ATH10K_PACKAGES = {}
-if env.GLUON_WLAN_MESH == 'ibss' then
-	ATH10K_PACKAGES = {'-kmod-ath10k', 'kmod-ath10k-ct', '-ath10k-firmware-qca988x', 'ath10k-firmware-qca988x-ct'}
+if env.GLUON_WLAN_MESH == '11s' then
+	ATH10K_PACKAGES = {'kmod-ath10k', '-kmod-ath10k-ct', 'ath10k-firmware-qca988x', '-ath10k-firmware-qca988x-ct'}
 end
 
 
diff --git a/targets/ipq40xx b/targets/ipq40xx
index f3c3c33b..e7a67f75 100644
--- a/targets/ipq40xx
+++ b/targets/ipq40xx
@@ -1,19 +1,19 @@
 local ATH10K_PACKAGES_IPQ40XX = {}
-local ATH10K_PACKAGES_IPQ40XX_QCA9888 = {'ath10k-firmware-qca9888'}
-if env.GLUON_WLAN_MESH == 'ibss' then
+local ATH10K_PACKAGES_IPQ40XX_QCA9888 = {}
+if env.GLUON_WLAN_MESH == '11s' then
 	ATH10K_PACKAGES_IPQ40XX = {
-		'-kmod-ath10k',
-		'kmod-ath10k-ct',
-		'-ath10k-firmware-qca4019',
-		'ath10k-firmware-qca4019-ct',
+		'kmod-ath10k',
+		'-kmod-ath10k-ct',
+		'ath10k-firmware-qca4019',
+		'-ath10k-firmware-qca4019-ct',
 	}
 	ATH10K_PACKAGES_IPQ40XX_QCA9888 = {
-		'-kmod-ath10k',
-		'kmod-ath10k-ct',
-		'-ath10k-firmware-qca4019',
-		'ath10k-firmware-qca4019-ct',
-		'-ath10k-firmware-qca9888',
-		'ath10k-firmware-qca9888-ct',
+		'kmod-ath10k',
+		'-kmod-ath10k-ct',
+		'ath10k-firmware-qca4019',
+		'-ath10k-firmware-qca4019-ct',
+		'ath10k-firmware-qca9888',
+		'-ath10k-firmware-qca9888-ct',
 	}
 end
```

rotanid, 2020-05-06:
have you tried v2019.1 and v2019.1.1 ?
@adrianschmutzler had initially contributed the support for the device in #1696

adrianschmutzler, 2020-05-07:
Just from the side:
I don't know which OpenWrt version is used in the referenced Gluon version. However, with Freifunk Franken Firmware we always had flashing problem with openwrt-18.06 on that device, which I could not localize to a particular change in OpenWrt.
With 19.07, flashing seemed fine. Since we've bumped in the meantime, and found nothing obvious with the 18.06-based version, we never found the root cause.

Despite that, it's known that TP-Link used several different bootloaders for the same device with CPE210 v1.1 (leading to some perfectly flashable, but others soft-bricked during TFTP). Maybe they have come back to that nonsense on v3.

In your case, my first suspicion is the openwrt-version, though, and this was specific to the v3 for some reason ...

(OT: Note that early support of the CPE210 v2/v3 also had broken WiFi)

F1ndus, 2020-05-07:
@rotanid 
Yes, 2019.1.1 and 2019.1 also dont work

@adrianschmutzler 
dc8d5d3 is based on 19.07 and the device also failed to start on it, but I'll double check this version, maybe I did something wrong, somewhere between there and ba34150 it starts working, I was not sure which commit exactly because some of these did not compile.

Guess I have to find another Hardware Model of my Router then, luckily I still can return it.

adrianschmutzler, 2020-05-07:
I obviously cannot say anything about the Gluon implementation, though. Someone should check that as well, of course.
Note that my comment on CPE210 v1 bootloader explicitly affected TFTP flashing. So, with that one, if you had the wrong one, you would not have been able to flash vendor firmware back. I'm not sure whether looking for another hidden revision will really bring you forward on v3, though ...

RadioBBS, 2020-05-17:
In the community FFHB, the loop issue exists for cpe210v3 devices in 2019.1.1 with SN greater than  219A594000900 . Smaller serial numbers work.

rotanid, 2020-05-25:
has it been tested with plain OpenWrt, too? i'm pretty sure already, it's not a Gluon bug, but a vendor change or OpenWrt thing....

findus, 2020-05-26:
@rotanid I still have the device, I will test it again with different OpenWrt versions

rotanid, 2020-09-02:
this issue is fixed with OpenWrt 19.07 commit fba22d89a918fa0e7dfc457cb3a6f43a1e08a8ed
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=fba22d89a918fa0e7dfc457cb3a6f43a1e08a8ed
this commit is included in our releases v2020.1.4 and v2020.2.1

closed 2020-09-02 by rotanid