freifunk-gluon/gluon #2018: next: Bootloop in x86-generic image

lemoer, 2020-05-09:
### Bug report

```
[    0.000000] Linux version 5.4.35 (lemoer@orange) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r13099+5-1553f58a98)) #0 SMP Mon Jan 6 02:42:15 2020
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000007fdffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000007fe0000-0x0000000007ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS ?-20191223_100556-anatol 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 1c4f001, primary cpu clock
[    0.000000] kvm-clock: using sched offset of 7947319693 cycles
[    0.000005] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000016] tsc: Detected 2494.220 MHz processor
[    0.003130] last_pfn = 0x7fe0 max_arch_pfn = 0x1000000
[    0.003182] x86/PAT: PAT not supported by CPU.
[    0.003189] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC  
[    0.007036] found SMP MP-table at [mem 0x000f5c90-0x000f5c9f]
[    0.007140] ACPI: Early table checksum verification disabled
[    0.007159] ACPI: RSDP 0x00000000000F5AC0 000014 (v00 BOCHS )
[    0.007167] ACPI: RSDT 0x0000000007FE156F 000030 (v01 BOCHS  BXPCRSDT 00000001 BXPC 00000001)
[    0.007173] ACPI: FACP 0x0000000007FE144B 000074 (v01 BOCHS  BXPCFACP 00000001 BXPC 00000001)
[    0.007181] ACPI: DSDT 0x0000000007FE0040 00140B (v01 BOCHS  BXPCDSDT 00000001 BXPC 00000001)
[    0.007184] ACPI: FACS 0x0000000007FE0000 000040
[    0.007187] ACPI: APIC 0x0000000007FE14BF 000078 (v01 BOCHS  BXPCAPIC 00000001 BXPC 00000001)
[    0.007191] ACPI: HPET 0x0000000007FE1537 000038 (v01 BOCHS  BXPCHPET 00000001 BXPC 00000001)
[    0.007228] 0MB HIGHMEM available.
[    0.007229] 127MB LOWMEM available.
[    0.007230]   mapped low ram: 0 - 07fe0000
[    0.007231]   low ram: 0 - 07fe0000
[    0.007239] Zone ranges:
[    0.007242]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.007244]   Normal   [mem 0x0000000001000000-0x0000000007fdffff]
[    0.007245]   HighMem  empty
[    0.007246] Movable zone start for each node
[    0.007246] Early memory node ranges
[    0.007247]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.007248]   node   0: [mem 0x0000000000100000-0x0000000007fdffff]
[    0.007250] Initmem setup node 0 [mem 0x0000000000001000-0x0000000007fdffff]
[    0.007764] Using APIC driver default
[    0.008121] ACPI: PM-Timer IO Port: 0x608
[    0.008134] ACPI: LAPIC_NMI (acpi_id[0xff] dfl dfl lint[0x1])
[    0.008179] IOAPIC[0]: apic_id 0, version 17, address 0xfec00000, GSI 0-23
[    0.008183] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.008186] ACPI: INT_SRC_OVR (bus 0 bus_irq 5 global_irq 5 high level)
[    0.008187] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
[    0.008193] ACPI: INT_SRC_OVR (bus 0 bus_irq 10 global_irq 10 high level)
[    0.008195] ACPI: INT_SRC_OVR (bus 0 bus_irq 11 global_irq 11 high level)
[    0.008204] Using ACPI (MADT) for SMP configuration information
[    0.008206] ACPI: HPET id: 0x8086a201 base: 0xfed00000
[    0.008224] smpboot: Allowing 1 CPUs, 0 hotplug CPUs
[    0.008249] [mem 0x08000000-0xfeffbfff] available for PCI devices
[    0.008250] Booting paravirtualized kernel on KVM
[    0.008253] clocksource: refined-jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645519600211568 ns
[    0.008261] setup_percpu: NR_CPUS:4 nr_cpumask_bits:4 nr_cpu_ids:1 nr_node_ids:1
[    0.008283] percpu: Embedded 28 pages/cpu s83084 r0 d31604 u114688
[    0.008321] KVM setup async PF for cpu 0
[    0.008328] kvm-stealtime: cpu 0, msr 7e8d400
[    0.008338] Built 1 zonelists, mobility grouping on.  Total pages: 32318
[    0.008340] Kernel command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=213c5684-02 rootwait console=tty0 console=ttyS0,38400n8 noinitrd
[    0.008520] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.008524] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[... 319 lines cut ...]
[    7.204227]  ? wait_woken+0x70/0x70
[    7.205658]  kthread+0xe6/0x100
[    7.207035]  ? kcompactd_do_work+0x1b0/0x1b0
[    7.208551]  ? kthread_park+0x90/0x90
[    7.210017]  ret_from_fork+0x2e/0x38
[    7.211447] Modules linked in: leds_gpio xhci_plat_hcd leds_apu2 fsl_mph_dr_of ehci_platform ehci_fsl act_police cls_basic sch_tbf sch_ingress gpio_button_hotplug button_hotplug tg3 ptp realtek pps_core mii libphy
[    7.216761] CR2: 00000000c809fefc
[    7.218162] ---[ end trace 64f309865abb407b ]---
[    7.219844] EIP: get_pfnblock_flags_mask+0x3d/0x50
[    7.221586] Code: 96 b6 c1 8b b0 2c 96 b6 c1 b8 1f 00 00 00 29 c8 81 e3 00 fc ff ff 89 c1 29 da c1 ea 0a c1 e2 02 89 d3 83 e2 1f c1 eb 05 29 d1 <8b> 04 9e 5b 5e d3 e8 23 45 08 5d c3 8d b4 26 00 00 00 00 55 89 e5
[    7.226739] EAX: 0000001c EBX: 0007ffff ECX: 0000000c EDX: 00000010
[    7.228759] ESI: c7e9ff00 EDI: c3b01f04 EBP: c3b01e54 ESP: c3b01e4c
[    7.230801] DS: 007b ES: 007b FS: 00d8 GS: 00e0 SS: 0068 EFLAGS: 00010206
[    7.232974] CR0: 80050033 CR2: c809fefc CR3: 0023c000 CR4: 000006f0
[    7.234987] Kernel panic - not syncing: Fatal exception
[    7.237042] Kernel Offset: disabled
[    7.238571] Rebooting in 1 seconds..
```

### Meta information

- [Link to site config](https://github.com/freifunkh/site/tree/1096d9e19a34dc5ed6c7ba6b982e86a295c71079)
- The image was started via qemu.
- Rebuilding the image did not help.
- It happens during the first boot.

txt-file, 2020-05-14:
@lemoer Can you please test if this is a gluon or an OpenWrt issue?

NeoRaider, 2020-05-30:
Is this still an issue? I can't reproduce it with the latest `next` branch.

NeoRaider, 2020-07-01:
@lemoer ping

lemoer, 2020-07-03:
Sorry for the late answer.

The issue seems to be solved in the current `next` branch. I'll close...

closed 2020-07-03 by lemoer