freifunk-gluon/gluon #2029: fastd occasionally segfaults in Gluon 2020.1.x

kpanic23, 2020-05-22:
### Bug report

**What is the problem?**
fastd crashes with segfault, causing the node to lose gateway connection.
```
kern.info kernel: [80946.000734] fastd[5014]: segfault at 10 ip 00007f3c229823e9 sp 00007ffd2ce68f60 error 6 in libc.so[7f3c2297f000+49000]
```

**What is the expected behaviour?**
- How do you think it should work instead?
fastd running without crashing :o)

- Did it work like that before?
Yes, up until 2019.1.2.

**Gluon Version:**
2020.1.2

**Site Configuration:**
https://github.com/ff3l/site-ff3l

**Custom patches:**
none

NeoRaider, 2020-05-22:
The `fastd-debug` branch of https://github.com/NeoRaider/gluon now contains a patch that should fix the issue (and add a few sanity checks to catch problems as soon as possible).

This will likely not be the fix that will eventually make it into upstream musl, as it introduces unneeded memory barriers in hot code paths even for completely single-threaded programs, but as a temporary fix to ensure correctness it should be sufficient.

NeoRaider, 2020-05-22:
I have replaced the fix with a better one that doesn't affect single-threaded programs.

NeoRaider, 2020-05-22:
And another update, this time with the full patch series proposed by Rich Felker. Please test!

oszilloskop, 2020-05-26:
Almost two days with two Futros with patched firmware there were no more fastd crashes.

NeoRaider, 2020-05-27:
Fixed with c3be82f6c82deb1777ad02ad2d09af5422489d37 and backported to v2020.1.x.

closed 2020-05-27 by NeoRaider