freifunk-gluon/gluon #2048: Gluon v2020.1.3: Missing entry in /etc/config/network after disabling outdoor option on outdoor device

mbaumga, 2020-06-06:
### Bug report

**What is the problem?**
The intial problem was that the 5GHz mesh0 interface was missing on the status page overview of a ubnt AC-Mesh-lite after I disabled the outdoor option in the configuration wizard. It didn't try to mesh with other devices either which were in the proximity. The 2.4GHz mesh1 interface worked just fine.
After some investigation it turned out that the mesh0 interface was existent, but in /etc/config/network the following entry was missing:
```
config interface 'mesh_radio0'
        option proto 'gluon_mesh'
```
Steps to reproduce:

- Have a freshly flashed/factory reset AC-Mesh-lite
- Go into configuration wizard (not advanced) and disable the default enabled outdoor option
- Add favourite ssh-keys, node name, location and contact information
- Save & restart device and let it configure itself for about 5 minutes
- Serve with internet uplink and juice
- Enjoy

**What is the expected behaviour?**
After disabling the outdoor option in the wizard, I expected the outdoor device to be configured like a non-outdoor device with 2.4 and 5GHz client and mesh interfaces on the default wireless channels, not the outdoor channels.

**Gluon Version:**
https://github.com/tecff/gluon/tree/v2020.1.x-tecff

**Site Configuration:**
https://github.com/tecff/site-ffa/blob/experimental/site.conf

**Firmware file**
http://fw1.tecff.de/v8/experimental/sysupgrade/gluon-tecff-v8-12-ge9f3a98-ubiquiti-unifi-ac-mesh-sysupgrade.bin

**Workaround:**
Current workaround is to add the missing entry in the /etc/config/network. After a reboot the device behaves as expected and has the 5GHz mesh0 interface working properly.
Since this isn't update safe, it's kinda hacky.

If there are things I can help to fix this issue, I am glad to be of service!

NeoRaider, 2020-06-06:
Alternative workaround: run `gluon-reconfigure`; any firmware update will also fix this. See #2049 for a proper fix.

closed 2020-06-07 by blocktrron

rotanid, 2020-06-11:
i just tested this, could reproduce the issue with v2019.1.2 and it improved with current v2020.1.x branch.
the section is not missing from /etc/config/network now when disabling outdoor usage in config mode.

but the 5ghz mesh interface still stays disabled, you have to manually enable it.
is this intended @NeoRaider and if yes, what's the reasoning?

rotanid, 2020-06-11:
> i just tested this, could reproduce the issue with v2019.1.2 and it improved with current v2020.1.x branch.
> the section is not missing from /etc/config/network now when disabling outdoor usage in [...]

reported this now in #2053

commit 600ab99f (2020-06-06): gluon-config-mode-outdoor, gluon-web-wifi-config: commit network config

200-wireless will add or remove the mesh network sections of
/etc/config/network. Commit this file, so the modified setting doesn't
get lost on reboot.

Fixes: #2048
PR #2049 (2020-06-07): Fix /etc/config/network commit for outdoor mode change, and related fixes
When looking into #2048 (fixed by the first commit of this PR), I noticed another issue in the outdoor mode setting: The advanced WLAN config page required an additional reload to update the mesh iface option visibility after changing the outdoor mode setting. This is fixed by the subsequent commits.

Fixes:  #2048
commit 8ecc5d89 (2020-06-07): gluon-config-mode-outdoor, gluon-web-wifi-config: commit network config

200-wireless will add or remove the mesh network sections of
/etc/config/network. Commit this file, so the modified setting doesn't
get lost on reboot.

Fixes: #2048
(cherry picked from commit 600ab99f809d78d6281ad999e7e4e4a91322a85c)
commit 5a17b819 (2020-06-07): gluon-config-mode-outdoor, gluon-web-wifi-config: commit network config

200-wireless will add or remove the mesh network sections of
/etc/config/network. Commit this file, so the modified setting doesn't
get lost on reboot.

Fixes: #2048
(cherry picked from commit 600ab99f809d78d6281ad999e7e4e4a91322a85c)