freifunk-gluon/gluon #2061: Assignment of MAC-generation indices

margau, 2020-06-22:
Hello,
the 8 MAC-addresses generated from 3bit-indices are completely used with the new OWE-feature, some already assigned multiple times (see https://gluon.readthedocs.io/en/latest/dev/mac_addresses.html).

In my opinion, in would be good to have an block of "user-assignable indices" for features like additional mesh/client-interfaces, and also a bit of space for possible additional features.
In https://github.com/freifunk-gluon/gluon/blob/2be6fe32e67ef167c2500c5f82ccb5ecbc6b1149/package/gluon-core/luasrc/usr/lib/lua/gluon/util.lua#L139 it says that the MACs is limited to 45bit-prefixes on the hardware-side. Is this relevant for all interfaces, or only for MAC's on Ethernet Interfaces?

Would it be feasible to extend the indice-range to 4 bit (at least), to allow for additional MAC's to be generated? Maybe it's possible to move the MAC's, which are not filtered out in Hardware, to the indices of the extended range.

What are opinions about this?

Related to discussion in #2059

blocktrron, 2020-06-22:
Possibly related to #1983

NeoRaider, 2020-06-22:
The table currently doesn't distinguish usage for batadv mesh and usage as a WLAN VIF. Each index can be used for both (where the interfaces `mesh0` and `mesh1` assume both roles at the same time). This makes the indices 2 and 6 free to be used for additional mesh interfaces like VLANs.

In https://github.com/freifunk-gluon/gluon/pull/1983#issuecomment-615838752 and https://github.com/freifunk-gluon/gluon/pull/1983#issuecomment-636187314 I proposed to make the assignment more dynamic, removing the need to specify the index explicitly, and to extend the MAC address range on demand.

In my opinion, this will be a prerequisite to make custom mesh interface support "official"; I might have a look into this next weekend.

In the mean time, users can use indices 2 and 6 as a temporary solution, but there is no guarantee that we won't reassign the indices at some point (like we did with the previous `ibss0`/`ibss1` indices).

belzebub40k, 2020-06-22:
> In the mean time, users can use indices 2 and 6 as a temporary solution, but there is no guarantee that we won't reassign the indices at some point (like we did with the previous `ibss0`/`ibss1` indices).

To not cause problems with future releases it might be better to use a MAC address from the locally administered range in the gluon_wired section. This is how we do it in our community. However I experienced that this does not work with all devices and sometimes you have to try several addresses before one works. I guess this is due to limits in the driver/chip which only accept MAC address from specific ranges.

NeoRaider, 2020-06-22:
@belzebub40k 
> To not cause problems with future releases it might be better to use a MAC address from the locally administered range in the gluon_wired section. This is how we do it in our community.

This is not possible for VXLAN interfaces, which are generated dynamically. The index is the only way to control the chosen MAC address for these interfaces, and only indices 0...7 are accepted. As multi-domain setups which require VXLAN are becoming more common, creating a proper way to support custom mesh interfaces has become a more pressing issue.

> However I experienced that this does not work with all devices and sometimes you have to try several addresses before one works. I guess this is due to limits in the driver/chip which only accept MAC address from specific ranges.

Such limitations exist only for WLAN adapters; the mesh addresses for wired/VXLAN interfaces are chosen from the same range solely to lower the probability address collisions between different nodes.

margau, 2020-06-23:
If the limitations are only for WLAN adapters, we can choose MAC adresses relatively free for Ethernet Interfaces? 
As custom interfaces are probably non-WLAN-interfaces, at least on the short term, I'd suggest to add another Range definition to #1983 which should be used only for "user-defined" interfaces. Would that be feasible @blocktrron?