freifunk-gluon/gluon #2063: Brief cross-domain link when switching domains

RalfJung, 2020-06-28:
### Bug report

To reproduce, make sure you have two nodes in the same domain that are meshing via wifi.
Then I run
```
uci set gluon.core.domain="<otherdomain>"
gluon-reconfigure
# wait a bit
reboot
```
on one of the devices.

**What is the problem?**
After "reconfigure" and before the reboot, router advertisements leak from one mesh to the other. So suddenly all nodes in one domain also have the IPv6 prefix from the other domain.

**What is the expected behaviour?**
Cross-domain links like this should be prevented.

**Gluon Version:**
Gluon 5696be72

**Site Configuration:**
https://git.hacksaar.de/FreifunkSaar/gluon-site

**Custom patches:**
none

blocktrron, 2020-06-28:
Just to be sure - this affects only the prefixes gluon-radvd advertises on the switching nodes?

RalfJung, 2020-06-28:
I am not sure -- we saw all 5 prefixes on the other side:
```
    inet6 2a03:2260:3009:1400:32b5:c2ff:fe6e:9d6b/64 scope global dynamic 
       valid_lft 1772sec preferred_lft 272sec
    inet6 2a03:2260:3009:1200:32b5:c2ff:fe6e:9d6b/64 scope global dynamic 
       valid_lft 1771sec preferred_lft 271sec
    inet6 2a03:2260:3009:1300:32b5:c2ff:fe6e:9d6b/64 scope global dynamic 
       valid_lft 1764sec preferred_lft 264sec
    inet6 2a03:2260:3009:1100:32b5:c2ff:fe6e:9d6b/64 scope global dynamic 
       valid_lft 1782sec preferred_lft 282sec
    inet6 fd4e:f2d7:88d2:fffd:32b5:c2ff:fe6e:9d6b/64 scope global dynamic 
       valid_lft 86364sec preferred_lft 14364sec
    inet6 2a03:2260:3009:400:32b5:c2ff:fe6e:9d6b/64 scope global dynamic 
       valid_lft 1792sec preferred_lft 292sec
    inet6 2a03:2260:3009:300:32b5:c2ff:fe6e:9d6b/64 scope global dynamic 
       valid_lft 1765sec preferred_lft 265sec
    inet6 2a03:2260:3009:200:32b5:c2ff:fe6e:9d6b/64 scope global dynamic 
       valid_lft 1738sec preferred_lft 238sec
    inet6 2a03:2260:3009:100:32b5:c2ff:fe6e:9d6b/64 scope global dynamic 
       valid_lft 1770sec preferred_lft 270sec
    inet6 fd4e:f2d7:88d2:ffff:32b5:c2ff:fe6e:9d6b/64 scope global dynamic 
       valid_lft 86370sec preferred_lft 14370sec
    inet6 fe80::32b5:c2ff:fe6e:9d6b/64 scope link 
       valid_lft forever preferred_lft forever
```
Here, `2a03:2260:3009:1x00` and `fd4e:f2d7:88d2:fffd` are the wrong addresses.

blocktrron, 2020-06-28:
`2a03:2260:3009:1000::/52` is contained in the extra prefixes of your site.

So the node itself does not advertise the prefixes,  the prefixes have to originate somewhere else. Do you have a connection to a gateway which advertises prefixes from this range?

NeoRaider, 2020-06-28:
Hmm, this is pretty much what I'd expect happening:
- some parts of Gluon will use the new settings as soon as `gluon.core.domain` is set
- some parts of Gluon will use the new settings as soon as `gluon-reconfigure` is run
- some services need a restart to use the new settings

All in all, switching domains at runtime (i.e. outside the config mode) is still not really supported...

One possible approach to fix this would be to implement a way to switch domains and reconfigure during reboot (and at `gluon-reload` time for the hoodselector).

RalfJung, 2020-06-28:
> So the node itself does not advertise the prefixes, the prefixes have to originate somewhere else. Do you have a connection to a gateway which advertises prefixes from this range?

Yes, those 4 prefixes are announced by our gateways. (I do not know if the nodes also announce those prefixes, but probably they do not.)

> One possible approach to fix this would be to implement a way to switch domains and reconfigure during reboot (and at gluon-reload time for the hoodselector).

Having a single command that apply the change and immediately reboots would probably already help, to at least keep the time down that the config is inconsistent. It could even shut down uradvd before applying any changes to prevent it from sending out announcements too early.

Does hoodselector not change domain at runtime?

NeoRaider, 2020-06-28:
> Having a single command that apply the change and immediately reboots would probably already help, to at least keep the time down that the config is inconsistent. It could even shut down uradvd [...]

The hoodselector uses `gluon-reload`; I assume it suffers from the same issue. I'm not sure what "support status" we actually claim for the hoodselector, it still feels somewhat "experimental" to me.

RalfJung, 2020-06-28:
> It could even shut down uradvd before applying any changes to prevent it from sending out announcements too early.

That doesn't help of course for the RAs from the other domain's gateways that are always crossing into the wrong domain. Maybe tunneldigger is also reconfigured and restarted already.

RalfJung, 2020-06-28:
> The hoodselector uses gluon-reload

Oh, the one that works so well that the docs got changed to recommend a reboot instead?^^

NeoRaider, 2020-06-28:
> Having a single command that apply the change and immediately reboots would probably already help, to at least keep the time down that the config is inconsistent. It could even shut down uradvd before applying any changes to prevent it from sending out announcements too early.

I propose to solve this porperly by only setting a flag for the intended switch during shutdown, and doing the actual switch during boot (before setting up the network).

RalfJung, 2020-06-28:
Sure, that would be much more work and much better.