freifunk-gluon/gluon #2064: 2018.2.4+2020.1.3 compatible Ubiquity AP and AP LR (long range) are not distinguished

smoe, 2020-06-30:
Separate versions of the firmware are offered for these two devies that are just symlinks for real:
```
lrwxrwxrwx ...    59 Jun 14 02:08 gluon-ffoh-0.9.21~exp20200614-ubiquiti-unifi-ap-lr-sysupgrade.bin -> gluon-ffoh-0.9.21~exp20200614-ubiquiti-unifi-sysupgrade.bin
```
On the map and the node info page, the device describes itself as
```
Geräte-Modell | Ubiquiti UniFi
```
which tends to confuse users, for whom the "LR" likely is important to distinguish their installed devices. And one is tempted to wander if the installed firmware was the correct one if not aware of the symlink.

Is there something that could be done about it? I likely can help with access to such a device if that is of any help.

Cheers,
Steffen

NeoRaider, 2020-06-30:
It's likely that the devices can be distinguished by some value in their EEPROM partition. Please provide us with dumps of the EEPROM partition (as listed in `/proc/mtd`) of at least 2 devices of each model (preferably more).

smoe, 2020-06-30:
We have 7 installed, but the only one I am confident about and have access to at the very moment is:
```
root@Carpe-Haus2-UG-west-K6-UbiquityAP_LR:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00760000 00010000 "firmware"
mtd3: 001c0000 00010000 "kernel"
mtd4: 005a0000 00010000 "rootfs"
mtd5: 00350000 00010000 "rootfs_data"
mtd6: 00040000 00010000 "cfg"
mtd7: 00010000 00010000 "EEPROM"
```
Another two LR are still boxed up. I hope to activate them temporarily over the upcoming days to provide this data. Others following this thread please help out for confirmed non-LR Unify APs.

NeoRaider, 2020-06-30:
@smoe Okay, so EEPROM is mtd7. You can just run `cat /dev/mtd7 > /tmp/eeprom` to read the contents of the partition into a file.

rotanid, 2020-06-30:
woah, old devices... we have only one of those in the field.
here's mtd7 eeprom from a non-LR non-AC "Unifi AP": https://upthis.de/dl/hnfcku/unifi_ap_eeprom
note: link is valid for only 7 days!

NeoRaider, 2020-07-01:
@smoe It looks like the UniFi AP uses the same EEPROM offset as the AirMax XM series. I've implemented the model detection in #2066

smoe, 2020-07-02:
@NeoRaider , Thank you for this. Reads great.

rotanid, 2020-08-08:
re-opening this issue as #2066 is not merged yet

closed 2024-12-23 by rotanid