freifunk-gluon/gluon #2070: Mark Ubiquiti Rocket M5 as Deprecated or Broken

oliver, 2020-07-03:
The German Bundesnetzagentur has announced that the "UBIQUITI ROCKET M5" must not be used within Europe any more. From https://www.bnetza-amtsblatt.de/download/36 section "Vfg Nr. 73/2020":

> Das weitere Bereitstellen, Inverkehrbringen und die Weitergabe des unten aufgeführten Gerätes im europäischen Markt wird untersagt.
> 
> Angaben zum Gerät:
> Gerätetyp: 	WLAN Basis-Station
> Modell:	UBIQUITI ROCKET M5
> Hersteller:	UBIQUITI NETWORKS INC., USA
> [...]
> Im Prüfbericht wurde darauf hingewiesen, dass die Grenzwerte der Störemissionen sowie für die Nebenaussendungen in den für das Gerät angegebenen Frequenzbereichen nicht eingehalten werden.

Note that this actually affects all of Europe (not just Germany); probably other countries have made similar proclamations.

Therefore I propose that this hardware is marked as deprecated or broken, so that the firmware is no longer built by default.

NeoRaider, 2020-07-03:
Hmm, the Rocket M2 and M5 share the same image. We currently can't deprecate only a single alias of a device (and allowing to do so would further complicate our already very complex target definition handling), so I guess we should just remove the alias? (turning it into a manifest_alias to make sure existing installations can still get updates)

SvenRoederer, 2020-07-03:
> The German Bundesnetzagentur has announced that the "UBIQUITI ROCKET M5" must not be used within Europe any more.

Is it really "must not be used anymore"? As I read it, it's not allowed to give it to somebody else (selling or even as a present). Even it's not written by the words it seems a logical consequence not to operate the device anymore. Probably this will announced in the next months.

NeoRaider, 2020-07-03:
I agree that the publication does not prohibit the operation alone of these devices. But regardless of the interpretation of the order, it is not in our interest to promote the operation of hardware that is known to interfere with other devices.

dangowrt, 2020-07-03:
The hardware allows for very high transmit power (>=1W) if configured with regulatory domains which allow that. I assume that some people have just abused it in that way and it became "famous" among "TX-power fanatics"... It may hence also very be that the device (if operating within legal ETSI power levels) is just fine.

SvenRoederer, 2020-07-03:
@dangowrt  the wording is: 

> Das Gerät wurde seitens des Mitgliedstaates einer Messung un-
terzogen. Im Prüfbericht wurde darauf hingewiesen, dass die
Grenzwerte der Störemissionen sowie für die Nebenaussendun-
gen in den für das Gerät angegebenen Frequenzbereichen nicht
eingehalten werden.

I assume that the french authority did correct measurements and used a conforming setup.

dangowrt, 2020-07-03:
> @dangowrt the wording is:
>
>
> I assume that the french authority did correct measurements and used a conforming setup.

I imagine them repeatedly confiscating that model emitting crazy transmit power and/or ignoring radar avoidance rules (supposedly setting wrong country code or by using 3rd-party-firmware).
Thing is that this was the most TX-power-strong 802.11an hardware (with RP-SMA plugs soldered on the board, otherwise Bullet-M5 is similar I guess) sold on the european market until now.

SvenRoederer, 2020-07-03:
They explicitly mentioned "suspicious emissions" and "adjacent channel interference" not something easy like "to high EPIR" or "not conforming to DFS-rules". The  2 latter would have made it much easier to regulate.

At the end of the order they put the official way to contact them and "blame" them.

dangowrt, 2020-07-03:
> They explicitly mentioned "suspicious emissions" and "adjacent channel interference" not something easy like "to high EPIR" or "not conforming to DFS-rules". The 2 latter would have made it much easier to regulate.

I just read that now and yes, that's better, because if it's really the hardware being noisy (or becoming noisy over the years) case ubnt is less likely to blame that incident on the use of 3rd-party-firmware and increase their locking efforts... (and in that case it's legitimate to argue that gluon, OpenWrt and so on should stop shipping binaries for that hardware)

tackin, 2020-07-04:
The authorities
- deny trading, importe and selling of that device
- explicit don't deny the use of of devices already sold and are out in the market

So if the further use of the device for a client is not illegal, why should we not deliver firmware updates?
Our firmware does not change/worsen the existing hardware problem. And that problem it self seems not that serious, that these devices have to be switchted off by law.

blocktrron, 2020-07-04:
@tackin 

@Neoraider wrote explicitly about turning the alias into a manifest_alias so that no image with the ROcket M5 name is generated but existing devices still get updates served.

PR #2127 (2020-10-06): ar71xx-generic: only create manifest alias for Rocket M5
This follow up the discussion done in #2070 by not  creating a symlink
for the Rocket M5. Images for the Rocket M2 can still be flashed on a
Rocket M5.

This change will prevent the Rocket M5 from appearing in Firmware
selectors. Existing devices will still receive updates, as the device
name is still referenced for the device name expected by the M5.

Closes #2070
commit 177ff68c (2020-10-06): ar71xx-generic: only create manifest alias for Rocket M5 (#2127)

* ar71xx-generic: only create manifest alias for Rocket M5

This follow up the discussion done in #2070 by not  creating a symlink
for the Rocket M5. Images for the Rocket M2 can still be flashed on a
Rocket M5.

This change will prevent the Rocket M5 from appearing in Firmware
selectors. Existing devices will still receive updates, as the device
name is still referenced for the device name expected by the M5.

Closes #2070

* docs: remove Rocket M5 from supported devices
closed 2020-10-06 by blocktrron