freifunk-gluon/gluon #2079: Hardware Support for TP-Link TL-WR802N v4

simonkurka, 2020-07-20:
Are there known reasons to not officially support _TP-Link TL-WR802N v4_ in Gluon?

According to https://openwrt.org/toh/tp-link/tl-wr802n_v4 there is OpenWrt-Support and it has a _MediaTek MT7628N_, 8MB Flash and 64MB RAM.

Thank you in advance!

blocktrron, 2020-07-20:
Someone in possession of the device has to confirm the device fully supports core gluon configuration (see https://github.com/freifunk-gluon/gluon/wiki/Device-Integration-checklist).

As the device is supported in OpenWrt 19.07, integration should be possible without any upstream patches.

simonkurka, 2020-07-20:
Ok, thank you!

I'll close this for now and may reopen it, if someone (maybe myself) has the possibility to test it.

closed 2020-07-20 by simonkurka