freifunk-gluon/gluon #2080: factory support for TP-Link CPE210 3.20

adrianschmutzler, 2020-07-20:
I just pushed an update to the supported devices of the CPE210 v3 to OpenWrt master, as flashing via GUI was not possible on v3.20:

https://github.com/openwrt/openwrt/commit/4a2380a1e778a8f8e0bfb0a00f2996ed0aab58d8

This will be backported to 19.07 later this week, the backport is just a trivial cherry-pick.

Support had been requested in OpenWrt and Freifunk forums recently.

This issue is meant as a "notification" in case you want to implement this for Gluon beyond just doing a bump of the OpenWrt version at some point.

If you are not interested or will just take it with the next OpenWrt bump, just close this issue. I'm not using Gluon myself and will not pursue this any further.

NeoRaider, 2020-08-15:
I think it makes sense to update our aliases accordingly.

Do I remember correctly that the CPE210/510 use the full revision number (including the 'v3.20') in `/tmp/sysinfo/model`?

mweinelt, 2020-08-15:
They do.

```
root@64283-cccda-s:~# cat /tmp/sysinfo/model 
TP-Link CPE210 v1.0
```

adschm, 2020-08-15:
Note that unless gluon changes /tmp/sysinfo/model locally, this will be gone with ath79, as there the DTS model will be taken.

rotanid, 2020-08-15:
reopen as #2096 needs to be backported, i think @NeoRaider will take care of this

rotanid, 2020-08-16:
backported via cherry-pick to v2020.1.x and v2020.2.x

PR #2096 (2020-08-15): New CPE210/... v3 versions strings
Clean up aliases (moving minor version strings to manifest_aliases) and add the new CPE210 3.1 and 3.20 strings to allow upgrades.

Closes: #2080
closed 2020-08-16 by rotanid