freifunk-gluon/gluon #2081: Unused network interface option "transitive"

ecsv, 2020-07-22:
More a cleanup ticket than a bug ticket

I've just checked gluon 2020.2 and master for the transitive option. I saw it in:

    config interface 'mesh_wan'
            option ifname 'br-wan'
            option index '0'
            option proto 'gluon_wired'
            option disabled '1'
            option transitive '1'
    
    config interface 'mesh_lan'
            option igmp_snooping '0'
            option ifname 'eth1'
            option index '4'
            option proto 'gluon_wired'
            option disabled '1'
            option transitive '1'
            option macaddr '0e:f0:78:5d:08:3c'
    
    config interface 'mesh_vpn'
            option ifname 'mesh-vpn'
            option transitive '1'
            option mtu '1426'
            option macaddr '0e:f0:78:5d:08:3f'
            option fixed_mtu '1'
            option proto 'gluon_mesh'

The only scripts which reference it are:

    lib/netifd/proto/gluon_mesh.sh: proto_config_add_boolean transitive
    lib/netifd/proto/gluon_mesh.sh: local fixed_mtu transitive
    lib/netifd/proto/gluon_mesh.sh: json_get_vars fixed_mtu transitive
    lib/netifd/proto/gluon_mesh.sh: export TRANSITIVE="${transitive:-0}"
    lib/netifd/proto/gluon_mesh.sh: json_add_boolean transitive "$TRANSITIVE"
    lib/netifd/proto/gluon_wired.sh:        proto_config_add_boolean transitive
    lib/netifd/proto/gluon_wired.sh:        local transitive index
    lib/netifd/proto/gluon_wired.sh:        json_get_vars transitive index
    lib/netifd/proto/gluon_wired.sh:        [ -n "$transitive" ] && json_add_boolean transitive "$transitive"
    lib/gluon/upgrade/500-mesh-vpn:transitive=true,
    lib/gluon/upgrade/220-interface-lan:if e:get('network','mesh_lan','transitive')==nil then
    lib/gluon/upgrade/220-interface-lan:e:set('network','mesh_lan','transitive',true)
    lib/gluon/upgrade/800-migrate-batadv:e:set('network',a,'transitive',t.mesh_no_rebroadcast)
    lib/gluon/upgrade/210-interface-wan:if e:get('network','mesh_wan','transitive')==nil then
    lib/gluon/upgrade/210-interface-wan:e:set('network','mesh_wan','transitive',true)

So it is is only translated to the environment variable TRANSITIVE. But no one seems to use it. In 2017, it was still used to set the (gluon specific) batman-adv option no_rebroadcast in package/gluon-mesh-batman-adv/luasrc/lib/gluon/mesh-batman-adv/config_mesh_interface. But this code is long gone

ecsv, 2020-07-22:
@T-X this is the stuff I mentioned during lunch. It most likely should have been dropped together with 2adf67edad86 ("Update routing packages (batman-adv v2016.5)"). But somebody else has to check if this has some other side effects which I am missing at the moment.

NeoRaider, 2020-07-22:
The option might still be useful for Babel - I believe we currently don't set the "split-horizon" setting correctly for all interfaces.

commit 79dc555b (2025-04-16): core: remove transitive option from network interfaces

closes #2081
PR #3485 (2025-04-16): core: remove transitive option from network interfaces
this removes an obsolete network interface config option "transitive"

i'm not sure, if the line in gluon-mesh-vpn-wireguard is necessary though - this was not part of the discussion back in 2020 in #2081 as there was no wireguard in Gluon back then.

closes #2081
closed 2025-04-16 by neocturne