freifunk-gluon/gluon #2082: fastd.mesh_vpn.enabled to gluon.mesh_vpn.enabled migration corner cases

1977er, 2020-07-22:
### Bug report

**What is the problem?**
On migration from previous versions `fastd.mesh_vpn.enabled` might not get converted correctly to `gluon.mesh_vpn.enabled`.

Migration script might not handle `fastd.mesh_vpn.enabled='on'` or `fastd.mesh_vpn.enabled=true`

**What is the expected behaviour?**
Would be nice, if the script can handle such corner cases.

**Gluon Version:**
v2019.1.1 to v2020.1.3

**Site Configuration:**
https://github.com/freifunkh/site

**Custom patches:**

NeoRaider, 2020-07-23:
Regarding our IRC conversation: Thinking about this again, I wonder if we should change this after all - we would need to add support for the alternative boolean representations to both `simple-uci` and the migration script itself.

Our UCI wrapper `simple-uci` only handles the values 1 and 0, as all other representations of booleans in UCI were deprecated long ago. OpenWrt's initscript helpers still handle the old options (`1|on|true|yes|enabled`), but as these values can never appear in a config that was only modified by Gluon's upgrade scripts or the recommended UCI shell commands, it seems more like a misconfiguration that happened to work in the past by chance rather than something we should support deliberately.

1977er, 2020-07-24:
Aye.

closed 2020-07-24 by 1977er