freifunk-gluon/gluon #2089: build fails after recent OpenWrt bumps

rotanid, 2020-08-14:
**What is the problem?**

after the recent OpenWrt bumps the build of Gluon branches based on OpenWrt 19.07 fails with "Error 2", attached is the relevant log output
[20200813_gluon_openwrt1907_openvswitch_wireguard_build_error.txt](https://github.com/freifunk-gluon/gluon/files/5072057/20200813_gluon_openwrt1907_openvswitch_wireguard_build_error.txt)

builds based on OpenWrt 19.07 commit 73fecd36bfd2b9f92a2a20f38bdb73b4433dec3e work, builds with later commits up to today fail.

the problem might be that both openvswitch and wireguard try to define some kernel prandom_u32* functions as backports
https://github.com/openvswitch/ovs/blob/branch-2.11/datapath/linux/compat/include/linux/random.h#L11
https://git.zx2c4.com/wireguard-linux-compat/tree/src/compat/compat.h#n243

**Gluon Version:**
branches v2020.1.x , v2020.2.x, master

**Site Configuration:**
irrelevant, also happens with other site configurations

**workaround**
please find attached a patch that disables the build of the openvswitch package which results in a working Gluon build.
the same behaviour is expected - but untested - when disabling wireguard builds.
[0001-disable-build-of-openvswitch.patch.txt](https://github.com/freifunk-gluon/gluon/files/5072069/0001-disable-build-of-openvswitch.patch.txt)

maybe the two packages can be fixed in a way that they don't conflict anymore, @NeoRaider wants to take a look into the issue

NeoRaider, 2020-08-14:
I can't reproduce this, please attach a full log that also includes the configure steps of these packages.

NeoRaider, 2020-08-14:
Also, do you have any local patches? Branches v2020.1.x , v2020.2.x and master all refer to a OpenWrt version that is currently on Kernel 4.14.187. Your log shows 4.14.193, so it must be a newer 19.07 version.

rotanid, 2020-08-14:
> Also, do you have any local patches? Branches v2020.1.x , v2020.2.x and master all refer to an OpenWrt version that is currently on Kernel 4.14.187. Your log shows 4.14.193, so it must be a newer 19.07 version.

sure, i did another bump to try if this fixes the problem. i won't build again with the older state, i think you can believe me and @mweinelt that for us it failed also with the older state you can find currently on github.

i just didn't push my bump, because it didn't fix things up and so it's not important

NeoRaider, 2020-08-14:
- Breakage with kernel 4.14.187 was fixed with the latest update to openwrt/packages, which was not backported to v2020.1.x and v2020.2.x by accident
- OVS breaks with kernel 4.14.193 again. This is fixed in OVS upstream, but not in openwrt/packages yet. We can wait with our next OpenWrt bump until this is fixed to avoid the breakage.

closed 2020-08-14 by NeoRaider