freifunk-gluon/gluon #2117: Ubiquiti NanoStation M2 should be in the tiny target

rubo77, 2020-09-07:
I think this (and some more) should be moved into the tiny target, since it has only 4 MB memory

rotanid, 2020-09-07:
https://openwrt.org/toh/ubiquiti/nanostationm2
no, it has 8M

blocktrron, 2020-09-08:
Even if, we are only referencing the target in which the device is upstream.

closed 2020-09-08 by blocktrron

rubo77, 2020-09-18:
But it has a big warning 

   DO NOT BUY DEVICES WITH 4MB FLASH / 32MB RAM

so the reason is, the only 32MB RAM

shouldn't we add those devices to the tiny target then as well? Or maybe create a new target for those with 32MB RAM?

rotanid, 2020-09-18:
Gluon doesn't decide which device belongs to which target, as @blocktrron already wrote.

also, the warning meant always an AND not an OR

please raise the discussion at OpenWrt communication platforms if you want to change such things.

SvenRoederer, 2020-09-18:
Recently I came up with this suggestion to the OpenWrt Devel List also: http://lists.openwrt.org/pipermail/openwrt-devel/2020-September/031422.html