freifunk-gluon/gluon #2129: WDR-3600 configuration-boot-loop

nudelkiste, 2020-10-08:
### Bug report

in the last days I bought 4 WDR-3600 on eBay and installed our gluon-image to it:
https://mgmt.saar.freifunk.net/firmware/stable/sysupgrade/gluon-ffsaar-1.9.0-tp-link-tl-wdr3600-v1-sysupgrade.bin
One out of the four WDR-3600 came with openwrt, and this one did the migration to gluon without any trouble.

**What is the problem?**
The three others came with stock FW. After installing our gluon images to it they bootet with the standard configuration screen. So far, so good.
But after configuring them, they did not reboot to the normal operation mode, but came up with the configuration screen again, and again, and again ...
It looked like they never where able to store their configuration or like they where reset somehow again and again.

**What is the expected behaviour?**
Normally they should enter standard operation mode as soon as the configuration is done.

**Workaround**
The only way out was installing the gluon version 1.8.1
https://mgmt.saar.freifunk.net/firmware/1.8.1/sysupgrade/gluon-ffsaar-1.8.1-tp-link-tl-wdr3600-v1-sysupgrade.bin
Doing so enabled me to store the configuration and to boot to standard operation mode.
Afterwards I could do an upgrade to 1.9.0 whiteout any trouble.
Just installing 1.9.0 directly did not allow me to switch from config mode to normal operation.

**Gluon Version:**
1.9.0 / gluon-v2020.2 -> unable to finish config mode when installing on WDR-3600 with stock FW, reboot always led to config mode again.
1.8.1 / gluon-v2019.1.2 used as workaround with upgrading to 1.9.0 / gluon-v2020.2 afterwards

**Site Configuration:**
https://git.hacksaar.de/FreifunkSaar/gluon-site

**Custom patches:**
none

rotanid, 2020-10-08:
which stock firmware version was running on the devices? knowing this, one can maybe reproduce your issue.

have you tried to flash factory OpenWrt image before trying the older Gluon version? this would allow to say if it's a Gluon-specific or OpenWrt upstream problem.

nudelkiste, 2020-10-08:
no, sorry, didn't try that.
From stock FW to 1.9.0 caused the config-reboot-loop. Installing in the config screen of the 1.9.0 the 1.8.1, reboot, configure and then it came up to normal operation mode. upgrade to 1.9.0 and I was on that stage I wanted to come to.
Coming from openwrt I could install 1.9.0 and configure it in one single step without any trouble.
At this time I did not note down the version of the openwrt as I did not expect this might become useful lateron. I wanted to go gluon ;-)

blocktrron, 2020-10-08:
First of all, what Gluon version is the "1.9.0" of your community? It's most likely visible on the map (vYYYY.XX)

>  Just installing 1.9.0 directly did not allow me to switch from config mode to normal operation.

Just to clarify - flashing 1.9.0 over an already installed 1.9.0 didn't mitigate the issue?

RalfJung, 2020-10-08:
Our 1.9.0 is based on v2020.2;
1.8.1 is based on 2019.1.2.

nudelkiste, 2020-10-08:
> Just to clarify - flashing 1.9.0 over an already installed 1.9.0 didn't mitigate the issue?

When I saw the config page of 1.9.0 (v2020.2) coming back again and again, I did not had the reflex to reinstall it with the same version. As I made good experience with 1.8.1 (2019.1.2) I took that one, ... and it worked to get out, somehow.

nudelkiste, 2020-10-09:
well, I tried to reprduce the behavior to figure, but failed.

1. tried "firstboot" and reconfiguring the node expeting to reenter the config-boot-loop, it simply worked like a charme without trouble.
2. went back to stock FW wdr3600v1_en_3_14_1 (150302):
`login as: root

BusyBox v1.30.1 () built-in shell (ash)

  _____  _____
_/ ____\/ ____\___________  _____ _______
\   __\\   __\/  ___/\__  \ \__  \\_  ___\
 |  |   |  |  \___ \  / __ \_/ __ \|  |
 |__|   |__| /_____/ (_____/(_____//__|
 -----------------------------------------------------
 OpenWrt 19.07-SNAPSHOT, r11098+13-9cafcbe0bd
 gluon-ffsaar 1.9.0 / gluon v2020.2
 -----------------------------------------------------
 Gluon commandline administration reference:
 https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration
 -----------------------------------------------------
root@ffsaar-c04a0038db10:~# pwd
/root
root@ffsaar-c04a0038db10:~# cd /tmp/
root@ffsaar-c04a0038db10:/tmp# ls -la
drwxrwxrwt   12 root     root           340 Jun 13 19:31 .
drwxr-xr-x    1 root     root             0 Jan  1  1970 ..
drwx------    2 root     root           240 Jun 13 19:19 .uci
-rw-r--r--    1 root     root            27 Jun 13 19:19 TZ
-rw-r--r--    1 root     root             0 Jun 13 19:19 dhcp.leases
drwxr-xr-x    3 root     root            60 Jun 13 19:19 gluon
drwxr-xr-x    2 root     root           260 Jun 13 19:19 lock
drwxr-xr-x    2 root     root            80 Jun 13 19:19 log
lrwxrwxrwx    1 root     root            21 Jun 13 19:19 resolv.conf -> /tmp/res                                                     olv.conf.auto
-rw-r--r--    1 root     root             0 Jun 13 19:19 resolv.conf.auto
drwxr-xr-x    4 root     root            80 Jan  1  1970 root
drwxr-xr-x    2 root     root           140 Jun 13 19:19 run
drwxrwxrwt    2 root     root            40 Jan  1  1970 shm
drwxr-xr-x    2 root     root            60 Jun 13 19:19 state
drwxr-xr-x    2 root     root            80 Jan  1  1970 sysinfo
drwxr-xr-x    2 root     root            40 Jun 13 19:19 tmp
-rw-r--r--    1 root     root       8060928 Oct  9  2020 wdr3600v1_en_3_14_1_sysupgrade.bin
root@ffsaar-c04a0038db10:/tmp# sysupgrade wdr3600v1_en_3_14_1_sysupgrade.bin
Image metadata not found
Saving config files...
Commencing upgrade. Closing all shell sessions.
`
3. flashed gluon to the node again
[image]
4. reconfigured the node as usual
[image]

and it rebooted to normal operation boot. 
I am really sorry, but I can´t reproduce it that way. Whenever I´ll flash the next one from scratch I´ll document the process better and report it here in case of trouble.

rotanid, 2020-10-09:
there is an easy workaround to the issue and it won't happen often anymore that someone factory-flashes such an old device.
at least as long as we can't reproduce it, this is a close.

closed 2020-10-09 by rotanid

NeoRaider, 2020-10-16:
@nudelkiste If you have any more details on how to reproduce this issue, I'll happily look into this, as I have such a device lying around.

nudelkiste, 2020-10-17:
Here the steps I did:
1. Connect the device with original FW to my local Network and connect to it (don't remember http://192.168.0.1 or http://192.168.1.1)
1. goto firmware upgrade and installed the latest version of our local firmware https://mgmt.saar.freifunk.net/firmware/stable/sysupgrade/gluon-ffsaar-1.9.0-tp-link-tl-wdr3600-v1-sysupgrade.bin
1. waited for the box to reboot
1. connected to the configuration page by http://192.168.1.1
1. configured first the extended part (2 ssh-keys, local WLAN with fixed full power for both, no outside option for 5G and no auto update for the software
1. configured then the main part (added mail as contact, no location, no bandwidth limit)
1. saved it and let the box reboot
1. wondered why the WLAN LED did not start flashing and realized that the configuration page came back as if I would just start the first time after flashing the box. Looked like none of the configuration I just did was stored, or if they had been erased during reboot again.

RalfJung, 2021-05-02:
> at least as long as we can't reproduce it, this is a close.

The issue seems to still arise with Gluon v2020.2.2; instruction to reproduce the problem have been given above. @NeoRaider did you have any luck reproducing the problem?

blocktrron, 2021-05-02:
@RalfJung How does dmesg look like on a node in config mode showing this problem? I was not able to reproduce the problem with my WDR3600.

RalfJung, 2021-05-02:
(I didn't have the problem, it was reported by @nudelkiste, a member of our Freifunk Saar community.)

nudelkiste, 2021-05-02:
when the node is coming up to config-mode, I add the ssh-key. But as it does not boot to normal mode but to config mode again, I have no idea how to login via ssh and how to get the dmesg output.
Otherwise I'll definitely try the next time. With the WDR3600 bought on eBay, I have this problem at 1/3 of the nodes, I'd say.

blocktrron, 2021-05-02:
> have no idea how to login via ssh and how to get the dmesg output.

In Config-mode the nodes have a password-less SSH login as root.