freifunk-gluon/gluon #2138: Gluon v2020.1.5 Tracking Issue

rotanid, 2020-10-19:
after about 1.5 months it's time for the last point release of the v2020.1.x branch.
per our Gluon Release life cycle, support ends with this final release: https://github.com/freifunk-gluon/gluon/wiki/Release-life-cycle

TODO: UPDATE THIS: we've backported a few fixes from v2020.2.x and updated it's OpenWrt 19.07 base.

Release Notes: https://md.darmstadt.ccc.de/gluon-v2020.1.5

NeoRaider, 2020-10-19:
Seems unnecessary to me. v2020.1.x is already past its EOL per https://github.com/freifunk-gluon/gluon/wiki/Release-life-cycle , and AFAIK there are no known regressions in v2020.2.x that could justify another v2020.1.x release.

rotanid, 2020-10-19:
@NeoRaider so i updated the issue to what the majority of maintainers voted for on IRC.

rotanid, 2020-11-01:
```
[ANNOUNCE] Gluon v2020.1.x End of life

The Gluon v2020.1.x branch has reached its end of life.
We have stopped maintaining it and will not create a release based on this branch anymore.

Updating to the v2020.2.x release series is the recommended course of action, which should be fairly easy. We are currently unaware of any regressions.

You can find documentation about the release life cycle of the Gluon branches on the GitHub Wiki:
https://github.com/freifunk-gluon/gluon/wiki/Release-life-cycle

```

closed 2020-11-01 by rotanid