freifunk-gluon/gluon #2143: Gluon v2020.2.2 Tracking Issue

rotanid, 2020-11-01:
after about two months it's time for the next point release of the v2020.2.x branch.
we've backported a few fixes from master and updated it's OpenWrt 19.07 base.

We could bump to the latest state again and backport #2133

Release Notes: https://md.darmstadt.ccc.de/gluon-v2020.2.2

yanosz, 2020-11-18:
Apparently, some checksum were changed upstream. This causes 2020.2.x to build fine and 2020.2.1 to fails as follows as of now:

```
mkdir -p /home/jan/src/gluon-build/gluon/openwrt/dl
SHELL= flock /home/jan/src/gluon-build/gluon/openwrt/tmp/.QCA99X0-firmware-5-ct-full-community-12.bin-lede.013.flock -c '       /home/jan/src/gluon-build/gluon/openwrt/scripts/download.pl "/home/jan/src/gluon-build/gluon/openwrt/dl" "QCA99X0-firmware-5-ct-full-community-12.bin-lede.013" "6fa74a3fc87cba97dbc4a7213b760f8d997cd9c5f11900d47d387b23764cf20a" "firmware-5-ct-full-community-12.bin-lede.013" "https://www.candelatech.com/downloads/ath10k-10-4b"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/firmware-5-ct-full-community-12.bin-lede.013
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://www.candelatech.com/downloads/ath10k-10-4b/firmware-5-ct-full-community-12.bin-lede.013
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  548k  100  548k    0     0   365k      0  0:00:01  0:00:01 --:--:--  365k
Hash of the downloaded file does not match (file: 6a95c4323e008499af8d3549513c8a128c69b97296888a4c4b642e6c14a65508, requested: 6fa74a3fc87cba97dbc4a7213b760f8d997cd9c5f11900d47d387b23764cf20a) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/firmware-5-ct-full-community-12.bin-lede.013
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/firmware-5-ct-full-community-12.bin-lede.013
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[4]: *** [Makefile:103: /home/jan/src/gluon-build/gluon/openwrt/dl/QCA99X0-firmware-5-ct-full-community-12.bin-lede.013] Error 2

```

blocktrron, 2020-11-18:
@yanosz There was a hard-disk crash at Candelatech which resulted in some firmware getting lost. OpenWrt has mirrors for that case, however this fallback fails in this version due to the firmware filenames not matching the one OpenWrt renames them into.

This is not really actionable for us (at least for v2020.2.1). v2020.2.x uses a different firmware which is still on their servers.

I've sent a fix for the download script to the OpenWrt list. We should consider backporting this to v2020.2.x for the next release in order to prevent this from happening in the future (https://patchwork.ozlabs.org/project/openwrt/patch/20201118152414.35309-1-mail@david-bauer.net/)

yanosz, 2020-11-18:
hmm.. I'm not sure, if `Hash of the downloaded file does not match (file: 6a95c4323e008499af8d3549513c8a128c69b97296888a4c4b642e6c14a65508, requested: 6fa74a3fc87cba97dbc4a7213b760f8d997cd9c5f11900d47d387b23764cf20a) - deleting download.` is driven by differences in filenames. Typically, hashes solely consider the content. 

IMHO its worth hightlighting fixed build problems in the release nodes of gluon 2020.2.2 - nevertheless, this solely regards OpenWRT.

blocktrron, 2020-11-18:
@yanosz Filename relates to the one used to retrieve the firmware / BDF from the OpenWrt mirrors.

Example:
https://www.candelatech.com/downloads/ath10k-10-4b/firmware-5-ct-full-community-12.bin-lede.013

this file is mirrored at

http://sources.openwrt.org/QCA99X0-firmware-5-ct-full-community-12.bin-lede.013

However, due to OpenWrt only querying with the filename from the original source (and not the one it is renamed to after downloading), downloading the firmware from the mirror fails.

See how it is trying to fetch

https://sources.openwrt.org/firmware-5-ct-full-community-12.bin-lede.013

This is not easily solvable for the tagged release, however this bug should be fixed in the future so we have a working fallback.

yanosz, 2020-11-19:
> This is not easily solvable for the tagged release, however this bug should be fixed in the future so we have a working fallback.

IMHO some docs / errata is good enough. For me, fixing would result in documenting build problems on demand.

RalfJung, 2020-11-20:
So, this issue breaks building all still-supported Gluon versions unless one already has a "hot" build cache that contains the required files... right? At least, a community I am helping out is currently unable to build a firmware, and the error message indicates that this is the problem they are having.

This would also mean that it is currently impossible to release a new firmware version for any community that uses CI-powered builds which start with a clean checkout each time. (I have not verified this though.)

NeoRaider, 2020-11-20:
Building using the latest v2020.2.x branch would work. We should try to get the v2020.2.2 release out as soon as possible - there is a number of bugfixes that are waiting for the release, including the fix for CVE-2020-27638.

RalfJung, 2020-11-20:
Ah, I guess for the experimental branch that's "good enough", fair (as long as doing auto-updates from the latest v2020.2.x the the eventual v2020.2.2 release will work, which I assume it does).

yanosz, 2020-11-20:
> So, this issue breaks building all still-supported Gluon versions unless one already has a "hot" build cache that contains the required files... right? At least, a community I am helping out is [...]

IMHO, 2019.1.3 is fine atm.

Greetz, yanosz

rotanid, 2020-12-07:
FYI: the current state (new commits were made today) of the branch is release-ready apart from one commit @blocktrron wants to include which is "related to iOS tethering"

Steini-, 2020-12-15:
I have to add, the download issue is still there. I'm still getting the 404s on several files, when building a new gluon-based firmware from scratch.
So please come on, fix the problem or try to explain to a total firmware building newbie why broken downloads are still a problem. I mean this issue was opened on 2nd of november, why does it take 6 weeks to fix broken downloads/sources?

rotanid, 2020-12-15:
v2020.2.2 was released on 2020-12-11, hence closing this issue.

rotanid, 2020-12-15:
@Steini- 

> I have to add, the download issue is still there. I'm still getting the 404s on several files, when building a new gluon-based firmware from scratch.
> So please come on, fix the problem or try to [...]

i built a new from scratch one yesterday and had no problems. please make sure you really use the newest commit and open a detailed bug report.

closed 2020-12-15 by rotanid