freifunk-gluon/gluon #2150: gluon v20.2.1: factory image is not accepted by Ubiquiti Nanostation M5

pullmoll, 2020-11-21:
### Bug report

**What is the problem?**
My new *Ubiquiti Nanostation M5* (NSM5) does not accept the factory image built for [ffkbu](https://github.com/johnnybee/ffkbu) - `wireguard` branch. I tried both, the [published image](https://images.ffkbu.de/BonnV2/stable/Wireguard/factory/gluon-ffkbu-V2-v2020.2.1-Wireguard-ubiquiti-nanostation-m5.bin) and also one I built locally.

The error message in the WebUI of the NSM5 reads `Please check that you have selected the correct firmware architecture for your device`.

**What is the expected behaviour?**
Factory image should be usable.

**Gluon Version:**
v2020.2.1

**Site Configuration:**
https://github.com/johnnybee/ffkbu branch `wireguard`.

**Custom patches:**
Don't know if ffkbu has any.

**Additional information**
The type label on the device says `M/N: NMS5`, `FCC ID: SWX-MSN`, and `IC: 6545A-MSN`.
I did log in to the device with `ssh` and tried to collect some perhaps relevant system information:

```
XW.v6.2.0# uname -a
Linux NanoStation M5 2.6.32.71 #1 Wed Jul 3 11:18:17 EEST 2019 mips GNU/Linux

XW.v6.2.0# cat /proc/cpuinfo 
system type		: Atheros AR934x
processor		: 0
cpu model		: MIPS 74Kc V4.12
BogoMIPS		: 266.24
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0000, 0x0888, 0x0c00, 0x0a08]
ASEs implemented	: mips16 dsp
shadow register sets	: 1
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

XW.v6.2.0# cat /proc/devices 
Character devices:
  1 mem
  4 /dev/vc/0
  4 tty
  4 ttyS
  5 /dev/tty
  5 /dev/console
  5 /dev/ptmx
  7 vcs
 10 misc
 13 input
 77 ATH_GPIOC
 90 mtd
108 ppp
127 gpio
128 ptm
136 pts

Block devices:
259 blkext
 31 mtdblock

XW.v6.2.0# cat /proc/iomem 
00000000-03ffffff : System RAM
  00002000-0020a5d7 : Kernel code
  0020a5d8-0028b81f : Kernel data
[... 312 lines cut ...]
ath_dev 297160 2 ubnt_poll,umac, Live 0x82c00000 (P)
ath_rate_atheros 111547 1 ath_dev, Live 0x83ae0000 (P)
ubnt_ewma_ht 15184 1 ath_rate_atheros, Live 0x839d0000 (P)
ath_hal 414872 3 ubnt_poll,umac,ath_dev, Live 0x83b80000 (P)
asf 6371 5 ubnt_poll,umac,ath_dev,ubnt_ewma_ht,ath_hal, Live 0x83a6a000 (P)
adf 7997 3 umac,ath_dev,ath_hal, Live 0x83a08000
urd 37367 1 ath_hal, Live 0x83aa0000 (P)
x_tables 10270 7 ebt_mark,ebt_mark_m,ebt_arpnat,ebt_vlan,ebt_ip6,ebt_ip,ebtables, Live 0x83a64000
pppoe 8256 0 - Live 0x83a94000
pppox 1418 1 pppoe, Live 0x839fa000
ppp_mppe 5184 0 - Live 0x83a6c000
ppp_async 6592 0 - Live 0x83a92000
ppp_generic 20053 4 pppoe,pppox,ppp_mppe,ppp_async, Live 0x83a78000
slhc 4987 1 ppp_generic, Live 0x83a06000
crc_ccitt 1067 1 ppp_async, Live 0x83a05000
ar724x_eth 67978 0 - Live 0x83a80000
libcrc32c 631 0 - Live 0x839c8000
crc32c 2112 1 - Live 0x839cf000
sha1_generic 1456 0 - Live 0x83a0f000
michael_mic 1616 0 - Live 0x839f8000
md5 1504 0 - Live 0x83a0c000XW.v6.2.0# uname -a
Linux NanoStation M5 2.6.32.71 #1 Wed Jul 3 11:18:17 EEST 2019 mips GNU/Linux
```

Any kind of help or hint is appreciated.

mweinelt, 2020-11-21:
Please consider everything in this section.

https://openwrt.org/toh/ubiquiti/nanostationm5#installation

 If gluon v2020.2.1 does not work, can you try an OpenWrt master snapshot?

pullmoll, 2020-11-21:
Nothing works, not with the WebUI and neither with tftp. I always see the origina Ubiquite airMax again.
I did a tiny script to get the sequence of commands sent fast enough, yet this also didn't help.
There is no v5.5 avaialable from ui.com. The earliest is 5.6.15 and also that image cannot be flashed through the WebUI.

It seems I had better not bought this thing.

NeoRaider, 2020-11-21:
You have a Nanostation XW, the correct image to use is `ubiquiti-nanostation-m5-xw`.

pullmoll, 2020-11-21:
I tried the ...m-xw... images, yes.

pullmoll, 2020-11-21:
Sorry, my bad, I did not follow the tftp procedur close enough. Without power cycling it doesn't work, i.e. you really have to press the reset pin and then power on the device. A small but important difference. Now going to try gluon-v2020.2.1.

pullmoll, 2020-11-21:
Ok, now I got gluon installed. Thanks for the help + closing this as it is a non-issue but the usual problem which sits 30 cm in front the display :-P

closed 2020-11-21 by pullmoll