freifunk-gluon/gluon #2155: Multiple B.A.T.M.A.N. patches for better performance

awlx, 2020-11-29:
As @FreifunkMUC we tracked down several issues of Batman when it runs in another encapsulation (eg VXLAN). The Batman project now has several patches for that. What is the best way to get them to Gluon?

Background:
https://gist.github.com/awlx/27c383511182f9bfb9d9e098abb823fe

Patches:
https://patchwork.open-mesh.org/project/b.a.t.m.a.n./patch/20201127173849.19208-2-sw@simonwunderlich.de/

mweinelt, 2020-11-29:
They should honestly come via the routing feed. CCing @ecsv for his opinion.

rotanid, 2020-11-29:
as far as i can see, the patches aren't even in upstream batman-adv yet?
(and i guess the patches are for batman-adv, not the protocol B.A.T.M.A.N. itself?)

awlx, 2020-11-29:
They got pulled into net today as far as I can see @rotanid.
https://lists.open-mesh.org/mailman3/hyperkitty/list/b.a.t.m.a.n@lists.open-mesh.org/message/Y7FYBSSUJO4MVDEDJBMBVNFRXO7TWLP6/

rotanid, 2020-11-29:
ah right sorry, 27 hours ago in upstream at git.open-mesh.org

awlx, 2020-11-29:
Also I am not saying this should happen _right_ now. I just wanted to open the issue to keep track of the changes :).

rotanid, 2020-11-29:
as it seems to be improvement, not bugfixing, i'd say we wait for the batman-adv release including these patches, then they will also be part of openwrt-routing and we would get them anyway by bumping the routing feed :)
other maintainers may have other opinions :)

ecsv, 2020-11-30:
These patches will be backported on the next release (2021.0) and put into the openwrt-routing feed. If it is really urgent for someone then I can also ask Simon to merge an early PR with these changes.

But also keep in mind that you also have to fix vxlan:

* https://patchwork.ozlabs.org/project/netdev/patch/20201126125247.1047977-1-sven@narfation.org/
* https://patchwork.ozlabs.org/project/netdev/patch/20201126125247.1047977-2-sven@narfation.org/

CodeFetch, 2020-12-29:
We somehow did not really measure a big improvement with the patches.
You can find the patches for the Gluon master here:
https://github.com/freifunkh/site/tree/pr_wireguard_vpn/patches

awlx, 2020-12-29:
Maybe it's not really measurable on the nodes :/.

CodeFetch, 2020-12-29:
@awlx Okay... I've thought we missed something. Wow I've realized we are working on it for a week now... It would be nice if you could join us in the Leinelab Mumble channel again. Here's the current state of the Gluon package:
https://github.com/lemoer/gluon/commits/pr_wireguard_vpn_rebased_on_master

It does not have a key submission API. That will be another package as we don't need automatic key submission.

awlx, 2020-12-31:
@CodeFetch we also implemented the patches in our latest firmware. I am still waiting for feedback if this helps on bigger offloaders etc.

awlx, 2021-01-01:
It seems these patches help the highly used offloaders to get slightly more throughput at the same load. So most likely the bottleneck on slower machines is something else already.

https://stats.ffmuc.net/d/hRIn3dRWk/mesh-nodes?orgId=1&var-nodeid=9ebdc859ce6e&from=1609282303000&to=now&refresh=5m

The offloader was upgraded 30.12 14:50

AiyionPrime, 2021-02-27:
> These patches will be backported on the next release (2021.0) and put into the openwrt-routing feed.

_ecsv_

Just wanted to leave a bump, that batman 2021.0 has been released about one month ago; https://www.open-mesh.org/news/100

And the routing feed got updated accordingly nine days ago.
https://github.com/freifunk-gluon/gluon/commit/e511b3bcb44f5a5ef54a5c101fa0cb5117e3e485

Which does contain https://github.com/openwrt-routing/packages/commit/6c81fc2916a56d7e655a2de6541437632e89792a.

So as soon as someone can confirm, 2021.0 does indeed contain these changes, we could move on from this, right?

ecsv, 2021-02-27:
@AiyionPrime It contains everything except the kernel VXLAN fixes (because the routing feed doesn't contain the kernel). And these patches were also backported to the openwrt-19.07 branch. The openwrt 21.02 and master branch contain the actual update to batman-adv 2021.0

rotanid, 2021-03-13:
i close this issue as there's not much to act on in Gluon.
Unless someone creates a PR to backport the relevant patches, Gluon will get it eventually via its base, OpenWrt

closed 2021-03-13 by rotanid