freifunk-gluon/gluon #2158: proper location of "targets/generic"

SvenRoederer, 2020-12-06:
During my work on PR #2142 I wonder if "targets/generic" should really belong into the "targets" folder?

It does not contain any target, in terms of board-types / CPU-types, but generic build-settings. By todays directory-layout the best place seems to be in the root-folder, as all other folders are as well matching as the targets folder.
So it's a bit the choice of polluting the root-folder or "misusing" the targets-folder.

CodeFetch, 2020-12-29:
These are the "generic" settings of all targets similar to the "generic" OpenWrt target which is the underlying Linux system. I agree this is confusing. It does not really belong there in OpenWrt, too: https://github.com/openwrt/openwrt/tree/master/target/linux/generic

rotanid, 2021-01-02:
assuming we move it to the root folder, it definitely needs another name though...

@NeoRaider what do you think?

NeoRaider, 2021-01-06:
I think the `targets` dir is fine for this, and I'd really like to keep the repo root clean.

The name `common` might be more accurate than `generic` (OpenEmbedded uses a `common` file for autoconf cache seeding that is not specific to an architecture), but I don't think it's important enough to change.

closed 2021-06-10 by rotanid