freifunk-gluon/gluon #2159: [doc, whishlist] Maintaining configuration when updating firmware

yanosz, 2020-12-07:
Considering maintaining a larger setup, I'm somewhat confused about how to maintain configuration data across firmware updates.

IMHO - the general OpenWRT approach is to create a tar-file including relevant data in /etc/config (among others). Hence, all configuration data is kept - the system continues with this configuration after an upgrade. For my impression, this does not hold for gluon, because upgrade scripts (c.f. https://gluon.readthedocs.io/en/latest/dev/upgrade.html) rewrite configuration to a large extent, which is not the case for OpenWRT.

Thus, I'm wondering: What is the strategy for Gluon to keep data, which is not purged on upgrades? What data is tar'ed (if any)? What happens to ssh-keys, geo locations, WLAN channel changes, additional interfaces?

IMHO, having https://gluon.readthedocs.io/en/latest/dev/upgrade.html only explains the control flow as of today, it'd good to outline the data flow during an update as well. 

I cannot make a suitable proposal, because I don't know about the actual rational.

lemoer, 2020-12-21:
Gluon just uses the normal OpenWrt mechanism for sysupgrades and therefore keeps the same things over the upgrade as OpenWrt in first place. However after this step, all scripts in */lib/gluon/upgrade/* are executed. These scripts alter the configuration in */etc/* in a meaningful way to provide upgrade strategies and changes to the site config.

yanosz, 2020-12-22:
> However after this step, all scripts in _/lib/gluon/upgrade/_ are executed. These scripts alter the configuration in _/etc/_ in a 
> meaningful way to provide upgrade strategies and changes to the site config.

Correct - and this ticket is about documenting

1.  Details about the "meaningful way" (e.g. how to preserve addtional interfaces)
2. Best-practices for meaningful ways in custom packages.

CodeFetch, 2020-12-29:
The reason that such a documentation does not exist, yet was that it is was unclear how to do it in the future. NeoRaider even thought about getting rid of UCI completely. But as far as I can see this won't happen and thus I agree a documentation would be nice... But can we please focus on the existing issues first? For example changing the WLAN channel. My first PR in Gluon was about that in 2014... It instantly frustrated me that this was denied. There are reasons for changing the WLAN channel... We should implement it before giving people advice how to do it manually... Just my two cents.

CodeFetch, 2021-01-01:
My comment was not off-topic.

PR #3530 (2025-06-30): docs: Add FAQ sections for setting retention and gluon_preserve
> At the GPN I learned a lot about others Gluon setups and how their settings would sometimes get lost after an upgrade, which was why they'd avoid them in the past.
> 
> This should clarify which settings are usually retained and migrated, and which are not.

This aims to resolve #2469.
It furthermore resolves #2159, unless @yanosz objects as I documented in that issue.

[This section](https://github.com/freifunk-gluon/gluon/pull/3530/files#diff-6eb68cf3015124e96821d08dd1dade74c4239aa8e2c39abad0d216698c4a289bR54-R58) is taken str [... cut]
closed 2025-06-30 by AiyionPrime