freifunk-gluon/gluon #2160: gluon-wan ssh uses mesh IPv6-Address.

yanosz, 2020-12-13:
### Bug report

**What is the problem?**
gl

**What is the expected behaviour?**

**Gluon Version:**

**Site Configuration:**

**Custom patches:**

rotanid, 2020-12-14:
maybe related to #1132 ? just guessing though

blocktrron, 2020-12-14:
Hmm, i don't see the issue here. `gluon-wan` only redirects DNS queries to the WAN DNS server and should not affect the routing decision.

The routing decision is based on a packet mark which is configured in the fastd config.

yanosz, 2020-12-14:
Hmm... I don't rember gluon-wan to be a DNS only thing. Afair it is designed as a general wrapper to execute arbitrary commands using the gluon-wan unix group. I had a discussion with @NeoRaider a few years ago on IRC, from which gluon-wan seemlingly evolved, but I don't remember many details. 

I don't know the packet marking process in detail. Afair it's possible to mark packets based on uid / gid using iptables and to apply to policy routing their ... but gluon firewalling somehow feels like a black box for me.

For me, it feels natural, to either

* To Implement policy routing based on the gid of a process
* Outline the DNS-specific case for gluon-wan by a proper documentation or even by renaming this tool to gluon-wan-dns.

blocktrron, 2020-12-14:
I've had a closer look on `gluon-wan` today. It's indeed only relevant for the DNS redirection to the wan dnsmasq. Thus closing here.

mweinelt, 2020-12-14:
Reopening as I believe we should clarify or extend the usage as requested in https://github.com/freifunk-gluon/gluon/issues/2160#issuecomment-744284482.

lemoer, 2020-12-21:
A posible and easy fix would be to rename it to `gluon-wan-dns`.

yanosz, 2020-12-27:
> A posible and easy fix would be to rename it to `gluon-wan-dns`.

Besides proposing exactly this before: I think the better way is a change in iptables, to match al traffic from gid=1000. However, I don't know why such a rule is not in place as of now, are there unwanted side-effects? (@NeoRaider)

PR #3547 (2025-07-19): mesh-vpn-core: rename gluon-wan to gluon-wan-dns
> to clarify its use-case

this resolves #2160 as suggested by @lemoer 

@neocturne Any objections?
closed 2025-07-19 by neocturne