freifunk-gluon/gluon #2161: busybox ip unable to show ip6 kernel routing tables

mweinelt, 2020-12-14:
### Bug report

**What is the problem?**

**What is the expected behaviour?**

All ipv6 routes in that table should be displayed.

Upgrading to `ip-full` the routes are displayed. Also they are visible on Gluon v2020.2 with the busybox `ip` we have there. So this is a regression.

**Gluon Version:**

Gluon v2020.2.2

**Site Configuration:**

n/a

**Custom patches:**

none

blocktrron, 2020-12-15:
Are you sure this worked on v2020.2? The relevant code in busybox seems to always return 0 when filtering ip6 tables:

https://git.busybox.net/busybox/tree/networking/libiproute/iproute.c#n104

mweinelt, 2020-12-17:
No, this was a machine that had `Ip-full` installed. So not a regression after all.

closed 2020-12-17 by mweinelt