freifunk-gluon/gluon #2164: question about gluon-core

dg0tm, 2020-12-28:
Hi,

i wonder if 001-upgrade try to acces an non exist file. i think it could/should try to access /lib/gluon/gluon-version instant?

https://github.com/freifunk-gluon/gluon/blob/049625bb14c845c94ac88e9d022d8e0273344577/package/gluon-core/luasrc/lib/gluon/upgrade/001-upgrade#L7

because

https://github.com/freifunk-gluon/gluon/blob/049625bb14c845c94ac88e9d022d8e0273344577/package/gluon-core/luasrc/lib/gluon/upgrade/999-version#L10 

use it...

blocktrron, 2020-12-28:
If I'm not mistaken, `/lib/gluon/version/core` was used ages ago to indicate the version prior upgrading (and is not used anymore).

`001-upgrade` therefore only sets the gluon_version non-nil, so initial steps are not executed again when upgrading. So the lack of `/lib/gluon/version/core` is perfectly fine.

closed 2020-12-28 by blocktrron

dg0tm, 2020-12-28:
Can you please rethink about it? Then without this file this script do nothing, because the first check is false and in these if clause mandatory. If the use case only set gluon_version non nil it could be also striped...

dg0tm, 2020-12-28:
And if i fixed it, the following scripts change their behavior, because the initial check are now funktional...

e.g. 030-system

blocktrron, 2020-12-29:
> Then without this file this script do nothing, because the first check is false and in these if clause mandatory.

This is expected, as "/lib/gluon/version/core" should only be available if you update from such an old gluon version. If you are not updating from such a version (which most likely won't work anymore anyway), this script is not necessary, as the prior Gluon version is already accessible using `gluon.sysconfig` and therefore the initial steps are skipped.

dg0tm, 2020-12-29:
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "
> "name": "View Issue"
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",

Okay, this script can be drop :)
Thank you for patience...